iNavFlight/inav #3306: FF_PIKOF4OSD weird motor output.

mrben1985, 2018-06-01:
### Board and Version
FF_PIKOF4OSD 
Version 1.9.1

### Behavior

Built a custom hex for 1.9.1 and have wired it all up. I am currently using crossfire and the motor is wired to S1. But it is responding to input from Pitch and Roll channels.

mrben1985, 2018-06-12:
I have tried the replacement target, however I am now getting servo output on motor 4/5 instead of 3/4 of the motors tab with the fixed wing mix. Is this something I have misconfigured or is the target still broken?

shellixyz, 2018-06-12:
Motor 4/5 ? This board has more than 4 motor outputs ? Could you tell me which motor pad is connected to which MCU pin ? I assumed they were right I only fixed the timers and added outputs for fixed wing.

mrben1985, 2018-06-12:
It has an S5 / S6 I do not have the board to hand, will check when I get home.

shellixyz, 2018-06-12:
Right, thanks. I don't know why they were commented as RC4/5 in the target I will change that.

fiam, 2018-06-22:
Fixed by #3333

PR #3333 (2018-06-15): Fix FF_PIKOF4 and FF_PIKOF4OSD targets
Closes #3306
closed 2018-06-22 by fiam