iNavFlight/inav #3309: Slow altitude gain in ALTHOLD mode.

NeilGaede, 2018-06-02:
In angle mode, steady hover, turn on ALTHOLD, neutral (spring to center) throttle. Copter jumps slightly, then steadily gains altitude. Blackbox shows a constant altitude gain on BARO. 

Flip off ALTHOLD, copter descends, flip on ALTHOLD, copter jumps slightly, then steadies and gains altitude steadily. 

What am I doing wrong. Do the velocity and/or position PID values have anything to do with it? 

Thanks for the help. Here is the dump and blackbox log. 

Neil 

[CLIDUMP1.txt](https://github.com/iNavFlight/inav/files/2064486/CLIDUMP1.txt)
[RISING.TXT](https://github.com/iNavFlight/inav/files/2064488/RISING.TXT)

NeilGaede, 2018-06-02:
I think I may have a possible answer. I have a spring to center throttle. My throttle_mid is set to 1500, and my deadband for altitude change is 50. Because I don't hover at 1500 I have my expo set about 25. When my throttle is at center blackbox show it at 1440 or so, which is above (below) deadband. So when I initiate althold, even though my throttle is in the middle, it see the expoed valve of 1440 instead of the actual middle position and somehow this messes up the velocity controller, though I'm not sure why it wouldn't descend instead of climb, but blackbox clearly shows a target velocity of 53 and target position of 360, though in theory I would think velocity target should be zero. 

Somehow it would be nice to be able to have nav_use_midthr_for_althold use middle stick position instead of middle rccommand_throttle.

NeilGaede, 2018-06-04:
After wading through the code I still don't have an answer to this issue. Is it possible it is just a position/velocity PID setting I need to make?

NeilGaede, 2018-06-07:
I think I may have a clue. I have mid_thr set to zero so I can get my expo right; if althold is using that value as middle throttle there will be a problem. I need a way to use mid_rc and middle throttle instead of my expo middle value. Not sure why you would use middle of expo as throttle center... 

I will try changing it in code and try it. Thanks for the tips though.

teckel12, 2018-06-07:
@NeilGaede See #3313

NeilGaede, 2018-06-08:
That is not good news. I just changed the code to use mid_rc instead of expo mid and it works perfectly.

NeilGaede, 2018-06-09:
I guess I could use nav_md_hover_thr or just hard code it to 1500. Either way, it would be nice to have the option without having to alter code and build myself.

digitalentity, 2018-06-09:
Actual problem is that throttle expo mid (percent) operates on range [minthrottle;maxthrottle] and is therefore not very intuitive because it's not directly related to stick travel. A proper solution would be to make throttle expo curve to operate on stick travel.

digitalentity, 2018-06-09:
Also note, that your actual throttle middle is not center stick. Stick travel is mapped by RC code from range [mincheck;2000] to [minthrottle;maxthrottle]. Since mincheck is 1100 by default, an exact middle of throttle range is at 1550 of raw stick value.

digitalentity, 2018-06-09:
There is an effort of heavy refactoring of rcCommand handling code which will eliminate the root cause of the issue, but I didn't have the time to finish it by 2.0.  Until it's finished - I'm afraid you'll have to maintain your own patched firmware (or figure out the throttle expo mid setting that will correspond to middle stick with your setup)

closed 2018-08-22 by stale[bot]