iNavFlight/inav #3315: Wing crash itself in RTH mode

flashez, 2018-06-03:
### Board and Version
INAV/OMNIBUS 1.9.1 Apr 21 2018 / 12:53:03 (03a5c19)
Actual board is Omnibus F3

### Behavior
The wing (Reptile S800) is dropping in Failsafe mode (RTH engaged).
Video compilation of actual wing stall: https://www.youtube.com/watch?v=0U8l5BLBX64
I suspect the throttle value of the RTH to be too low. If you have any suggestion :)

#### Blackbox logs/Config (.ZIP) where problem occurs
https://pastebin.com/1xHseGdx

shellixyz, 2018-06-03:
Yes looks like it stalls try to increase the throttle. Settings set `nav_fw_cruise_thr` and `nav_fw_pitch2thr`. Cruise throttle is the throttle it uses when level. `pitch2thr` is the amount the throttle is increased for each degree pitching up or decreased when pitching down.

flashez, 2018-06-04:
So @shellixyz do you mean I should increase `nav_fw_cruise_thr` but for `nav_fw_pitch2thr` I don't really know what I should do. I'm pretty new with inav and wings.

shellixyz, 2018-06-04:
You should set `nav_fw_cruise_thr` to the throttle value your wing is flying happily with when level. For `nav_fw_pitch2thr` fly at 10 degree up pitch and note the minimum throttle you can fly this way. Then set `nav_fw_pitch2thr` to (noted_thr - `nav_fw_cruise_thr`) * 0.11. You can set `nav_fw_max_thr` higher if you think it is too low it is the maximum throttle the FC will use when controlling the throttle automatically (like in RTH).

flashez, 2018-06-04:
Thanks a lot for the clear explanation @shellixyz
Lastly, how do you know can you see the throttle value (like on osd) ?
I have throttle % input only.

Tygrys-1, 2018-06-05:
Multiply throttle % by 10 and add 1000.

flashez, 2018-06-10:
So I made some changes since but the problem is still occuring.
Here is a DVR showing clearly the wing crashing right after RTH is engaged by failsafe:
https://www.youtube.com/watch?v=eZGbhM9_9bw

flashez, 2018-06-10:
[https://www.youtube.com/watch?v=eZGbhM9_9bw](url)

giacomo892, 2018-06-11:
Seems like your wing stalls as soon it banks. It is unable to enforce the max banking angle too. As soon it banks to turn toward home it ends in an loop. Are you sure surfaces have enough authority and GC is right?

flashez, 2018-06-11:
COG is ok but i don't really know what you mean by "surfaces have enough authority"

giacomo892, 2018-06-11:
Can you control the plane perfectly while you are in `ANGLE` mode?

flashez, 2018-06-11:
@giacomo892 yes without any problems, keeping it's altitude (ANGLE + ALT HOLD).
I've done several RTH before (RTH issued by failsafe, so fully automatic, not RTH mode) without any problems.

shellixyz, 2018-06-12:
> I've done several RTH before (RTH issued by failsafe, so fully automatic, not RTH mode) without any problems.

In your video it is the case (failsafe) and it still crashes. There is no difference between RTH on FS and RTH mode it is done by the same code. So if one is working the other should be working the same. Please provide a blackbox log for further analysis.

digitalentity, 2018-06-13:
@flashez if with same settings wing occasionally crashes when maintaining a turn - I'd blame CoG or insufficient airspeed. From your comments I can see that issue is intermittent which makes it impossible to tackle without knowing what FC is actually doing (logs)

closed 2018-08-26 by stale[bot]