iNavFlight/inav #3317: Communication with ROS

shlapion, 2018-06-04:
Greetings, 

I am using iNav in a project where I want to implement different CV algorithms, visual servoing for instance, on companion computers running ROS. 
One of the first steps I would like to take is to have the companion computer send instructions to to the FC during flight over the USB Port. 
The following question arose:

- Is it possible to send Roll/Pitch/Yaw commands to the FC during flight over the USB port (Basically mimicking the inputs form the RC). If so, is there any documentation on how one could proceed ?

As additional details, the hardware I am using are:

- Matek F405 OSD

- Raspberry Pi 3 
 

Thank you

DzikuVx, 2018-06-04:
Yes you can send MSP commands over any serial port. Probably not USB, but regular Serial from RPi to INAV. Looks for potential MSP libraries. Even those written for MultiWii or Cleanflight should work with the scope of rc commands

shlapion, 2018-06-05:
Thank you for the info @DzikuVx will look into it.

wx4cb, 2018-06-21:
i would look into flystick: https://github.com/jsa/flystick

shlapion, 2018-06-21:
Thx for the info @wx4cb 

For anyone who is also interested in this topic I found a ROS node which has a few interesting functionalities already built in. Am currently trying to write an ARM function. 
The [link](https://github.com/christianrauch/ros-multiwii) to the repo. 

Will return with infos once everything progress has been made

shlapion, 2018-07-09:
Looked into the issue deeper. I am able now to send MSP commands to the FC over USB, however i am not sure if they are processed. 
@DzikuVx could you point me in the right direction regarding the questions below:

1. I noticed that failsafe mode is activated if the RC is not turned on. I have set the `failsafe_procedure` to `NONE`. If i send MSP commands over the USB port, for instance set the appropriate channel values for ARMING, would the lack of an RC signal prevent the FC form arming? If so is there a way to bypass this?

2. Is there a way to set/send MSP messages using the Configurator GUI? If so how?

wx4cb, 2018-07-09:
@shlapion 

1. that's entirely possible, as it would either be waiting for a switch to arm or stick commands.

2. the only way would be via the cli.

closed 2018-09-21 by stale[bot]