iNavFlight/inav #3321: RSSI BY SBUS

Christianmix, 2018-06-05:
I have used a R9M module in my x9d for some time, and now I bought R9 mini receivers, and frsky's website says so: >>> With RSSI output in SBUS

My question is: Will it appear in the osd of my Omnibus F4 v3 the RSSI making only the connection by sbus? The receiver is about to arrive, there was a delay due to the transporters' strike in my country, but it's about to arrive in the next few days and I wanted to know how I will get RSSI on my screen. Thank you.

https://www.frsky-rc.com/product/r9-mini/

DzikuVx, 2018-06-05:
This probably means that RSSI is injected into one of not used channels and sent to FC. Then yes, it can be configured to be displayed on OSD

closed 2018-06-05 by DzikuVx

peecao, 2018-06-12:
I use the R9 Mini's in 2 of my quads.  I set up the RSSI using the sensors on my Taranis then adding an Input and finally a Mix using that input. 
Betaflight / iNav then, add the aux channel from the mix to the RSSI in the receiver tab and OSD

Christianmix, 2018-06-26:
@peecao  I did not understand how you did to have rssi, is it like this video? https://www.youtube.com/watch?v=t-evOAS9Mkg

teckel12, 2018-06-26:
@Christianmix That video sounds exactly what @peecao was saying.  I've done that as a test and it totally works.

Christianmix, 2018-06-27:
Sorry to not speak in the video, but my English is terrible, so I write here explaining what I did in the video.
I made the settings exactly as in the video, and it is not working, I turn the transmitter off and it keeps the rssi 100%, when I re-turn on the transmitter it takes a few moments to zero and returns to 100%, if I turn off the card and the receiver and relay with the transmitter turned off, stays at 50% as soon as I turn on the transmitter it goes to 100%, I can turn off as many times as I want the transmitter that does not drop again, only if I take the power off everything and reconnect it stays there 50% until I reconnect the transmitter.
https://www.youtube.com/watch?v=3HmIWJlQZ-4

@peecao   Is there anything different that you have done that is not in the video?

teckel12, 2018-06-27:
There's other tutorials as well on this, maybe another one would be more informative. I believe your issue is the mix. As I remember it needed to be adjusted quite a bit to work. Something like 50% and maybe even flipped. I don't remember, but it sounds like the mix is what you need to look at.

Christianmix, 2018-06-27:
This tutorial is correct, it is exactly like this, I was testing the wrong way, I discovered that I did the same test with an x8r and gave the same problem, ai thinking here alone I found out where I was wrong kkkkkkk I was turning off the contole to see if it would not happen rssi and this would not happen because it is not a voltage signal from the receiver to the FC but rather the control signal to the receiver and the receiver informs the FC through the indicated channel, I left my espoa in front of the computer with the telephone and I went walking with another phone and the rmm with 10mw, I walked some 60 meters outside my house and the signal lowered to 98% and she informed me that I had fallen, there I walked to 100 meters distant between house and cars and the signal went to 51 %, then since I was there I decided to put 500mw and the signal went up to 100% again. That is, the video is correct and works yes. Hugs and thanks for your attention.

teckel12, 2018-06-27:
@Christianmix Glad you got it sorted out, there's a lot of settings required to get it working correctly.

peecao, 2018-07-03:
Yes, that works perfectly, also works for the R9Mini as well.
*Sincerely,*
*Marty S.*
*0900932792*
*Line: peecao*