iNavFlight/inav #3322: Is there a screen offset setting for the OSD?

whakru, 2018-06-05:
matek f722
Is there a screen offset setting for the OSD?
As in betaflight: https://github.com/betaflight/betaflight/wiki/OSD-and-CMS-Adjusting-Screen
I have a menu that goes beyond the screen, which can be fixed by what comand?

shellixyz, 2018-06-05:
There is not. Could you post a screenshot ?

whakru, 2018-06-05:
@shellixyz 
Unfortunately at the moment I can not ..
I do not fit inscriptions in the menu on the width (Not all), on the right are cropped ..
And you can not see the menu title from above.
The modes of PAL NTSC do not change anything.
--------------------
К сожалению в данный момент не могу..
У меня не умещаются надписи в меню по ширине (Не все), Справа обрезаются..
И сверху не видно заголовок меню..
Режимы PAL NTSC ничего не меняют.

whakru, 2018-06-05:
@Tygrys-1 
What values can he have?

Tygrys-1, 2018-06-05:
I don't remember, but when you type in cli:
get osd_row_shiftdown 
you should have the range or list of allowed values, it works for all variables that way.

On my display the topmost row is half visible, so I put gps coordinates there. They are not visible in the gogles, but they are present on the recording on different receiver.

fiam, 2018-06-08:
If anyone has time, I think the best solution would be implementing the same settings as BF. It will be easier for users too.

whakru, 2018-06-08:
@fiam Yes, that's a good idea  ))

teckel12, 2018-08-07:
Bad bot!  This is something that's on my list of things to do.

teckel12, 2018-08-15:
Confirmed that `osd_row_shiftdown` is never used in 2.0 RC6.  Looks like it was started but never finished (or removed accidentally).  As the command already exists, maybe it should do something? 😉

gmacfpv, 2019-02-17:
I really need this feature. Need to move my whole OSD down a bit.

closed 2023-10-27 by DzikuVx