iNavFlight/inav #3323: Random disarm during FS with RTL on.

driesvandenbroecke, 2018-06-05:
### Board and Version
_Matek F405-OSD
_1.9.1

_INAV/MATEKF405OSD 1.9.1 Apr 21 2018 / 13:26:36 (03a5c1922)

Random disarm when I triggered RTL about 6.5km out. The plane dived vast to the fixed RTL height of 80 meters, which caused me to lose video and RF Signal. It came back at around 4 km, I still had bad image but I moved the sticks so it went from FS to RTL and then back to FS. I saw it was in FS flying home so I didn't touch anything anymore until I was for a very brief moment the stats on the screen. My aircraft got disarmed by itself.

video of the flight:  [https://youtu.be/fmgIClqqwgY](url)
dump: [https://pastebin.com/QmmiXJ5Q](url)

Tygrys-1, 2018-06-05:
On weak radio signal there can be false recognition of disarm switch.
See #3248, #3118

DzikuVx, 2018-06-06:
This issue is (or at least we very strongly belive so) will be fixed by INAV 2.0. Arming channel would have to go into "disarm" value for a period of time to trigger disarm

closed 2018-06-06 by DzikuVx