iNavFlight/inav #3325: No Work SDcard in Blackbox F722

whakru, 2018-06-05:
### Board and Version
INAV/MATEKF722 1.9.1 Apr 21 2018 / 13:43:00 (03a5c1922)
# resources

# mixer
mixer QUADX

mmix reset

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# feature
feature -RX_PPM
feature -VBAT
feature -TX_PROF_SEL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature -TRACE
feature VBAT
feature TX_PROF_SEL
feature TELEMETRY
feature CURRENT_METER
feature BLACKBOX
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
[... 505 lines cut ...]
set nav_fw_pos_xy_d = 8
set nav_fw_heading_p = 60
set thr_mid = 50
set thr_expo = 10
set tpa_rate = 0
set tpa_breakpoint = 1500
set fw_tpa_time_constant = 0
set rc_expo = 70
set rc_yaw_expo = 50
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set manual_rc_expo = 70
set manual_rc_yaw_expo = 50
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100

### Behavior
No Work SDcard in Blackbox F722..
I tried a lot of cards of different size and class ..
Tried to format and so on ..
I've already found out that many people do not see a memory card on f722.
on Betaflight all works ..
What to do?

lylus, 2018-06-05:
我也遇到了同样的问题 我以为是SD卡的问题 又重新购买了SD卡 看来只是一个bug

DzikuVx, 2018-06-06:
SanDisk 16GB Ultra card works. I know, I'm using them

lylus, 2018-06-06:
@DzikuVx OK, thank you. I have bought a SanDisk 16GB Ultra now, and I will test it after receiving it.

flyfisher2, 2018-06-06:
I am also using SanDisk Ultra 16Gb on two F722 boards, never had a problem with them.

digitalentity, 2018-11-04:
Fix ready, verified on several F7 boards. Closing.

closed 2018-11-04 by digitalentity