iNavFlight/inav #3335: Fixed Wing / RTH / Slowly spiralling down

flohoff, 2018-06-07:
Hi,
i am flying pretty well and i have RTH working basically. It returns and starts spiralling. But in a matter of minutes it looses height although land is set to never. 

I would have expected INav to consult the Baro for height but it seems it relies on the Fixed Wing
to be leveled perfectly once height it reached.

### Board and Version
OmniBus F4 Pro V3
Inav 1.9.1

flo@p4:~/projects/inav/inav-config$ grep rth config 
set nav_rth_allow_landing = NEVER
set nav_rth_alt_mode = FIXED
set nav_rth_altitude = 7000

Blackbox attached - At 3:03.193 i activate the RTH at ~63 Meters. RTH is set to 7000.
I let it spiral until 6:26.865 when the height dropped to ~40m (All Baro)

Flo

flohoff, 2018-06-07:
Here is the BlackBox log:
https://silicon-verl.de/home/flo/tmp/20180607-flo-rth-spiral-descend.zip

digitalentity, 2018-06-07:
Might be related to #3332

closed 2018-08-20 by stale[bot]