iNavFlight/inav #3341: Strange behavior of iNav 1.9.1 in PostHold

jstremmler, 2018-06-08:
I have a strange behavior of iNav 1.9.1 in PostHold mode with my Transtec Frog quad. 

My ESC's are BLHeli32 ones.

I have configured them to Multishot and have calibrated them carefully. 

When I am flying in Acro or Angle mode everything works fine. Motors run at approximately 28% load. 

Just by switching flight mode to PosHold, motor 1 ( and only motor 1) spins up to 85-100% load and gives a terrible noise like it would loose commutation. Motors 2, 3 and 4 stay where they are. 
Amperage in the Blackbox goes up from approximately 7A to 24 A!!!!

When I configure my ESC's to Oneshot125 everything works normal.

Attached is a Blackbox file which shows the behavior. 
(And believe me, this is an iNav issue, because motors run fine in Acro, Angle or Horizon mode...) 
[Frog11.TXT](https://github.com/iNavFlight/inav/files/2084598/Frog11.TXT)

DzikuVx, 2018-06-10:
I see very severe magnetometer issues
[image]

jstremmler, 2018-06-11:
Sure, because motor 1 is shaking and vibrating my quad like crazy. (If you look to the hardware design of a magnetometer then this is not a surprise)
But why is only motor 1 loosing commutation and running at 85 - 100% load just when hovering in PosHold?

And this does not explain why motor 1 behaves absolutely normal when I specify Oneshot 125 in the Configuration Menu and switch my copter to PosHold. And in that case the magnetometer also has no issues..... and the quad is staying in its position.

closed 2018-08-24 by stale[bot]