iNavFlight/inav #3343: INAV 2.0 disarm on low RSSI

driesvandenbroecke, 2018-06-08:
### Board and Version
_Board name goes here. If you have a link where you've purchased the board - please include it_
_2.0

# version
# INAV/MATEKF405OSD 2.0.0 Jun  6 2018 / 18:03:47 (f7c2824f)
_
### Behavior
Also a random disarm using 2.0 with openlrs on low rssi

[https://www.youtube.com/watch?v=nbJEz2IkpHw](url)

fiam, 2018-06-08:
@driesvandenbroecke Could you post your config dump? Do you have a blackbox log by any chance?

DzikuVx, 2018-06-09:
See here: https://youtu.be/nbJEz2IkpHw?t=31s

First model yanks to the right and only then disarms. I very strongly believe that this is due to receiver configuration/dehavior. It just seted arming channel is disarm value and done

Artyum, 2018-06-24:
I also experienced disarm during flight with lost of controls. I wonder, if in this case it was rx cause then it must generate wrong signals for over 0.5 sec for default inav configuration. If yes then maybe 0.5 sec is too low threshold.

closed 2018-09-06 by stale[bot]