iNavFlight/inav #3344: Bug iNav 1.91 + Matek F405-WING + R9M mini FPort + X9DP = no telemetry

davidngrc, 2018-06-08:
hardware: iNav 1.91 + Matek F405-WING + R9M mini FPort + X9DP
all firmware are in latest.
I connect the "R9M mini" smart port to "Matek F405-WING" Sbus port.
Then set "UART2" as SerialRX in INav Configurator 1.93 Ports tab. (did not set any SmartPort uart)
Then set "FPort" in INav Configurator "Receiver Mode" -> Serial-base receiver
If I move the X9DP stick, I can see the movement in Nav Configurator Receiver tab.
go to X9DP -> discover new sensor, only show two, RSSI and RxBat

I re-fresh to NON-fport firmware to "R9M mini"
I use the normal wiring sbus to sbus, smart port to smart port.
discover sensor, 14 of them show up...

I think there is a bug in iNav FPort handling.

P-I-Engineer, 2018-06-08:
the sbus pin on your f405 wing most likely does not have a bidirectional inverter. You'll need to find the inverted smartport signal on your R9 mini and hook that to a free TX port on your controller.
I will be attempting this tonight on a F405-ctr

davidngrc, 2018-06-09:
I did try to connect the TX2 (which is the softserial 1) to smart port on "R9M mini" , also didn't work.

davidngrc, 2018-06-09:
Just try TX6, also don't work.
try different serialrx_halfduplex and serialrx_inverted also don't work.

P-I-Engineer, 2018-06-09:
Are you using the s.port pinout on the r9 mini? You need to find the inverted pad for sport on the r9 mini, if it exists.

davidngrc, 2018-06-09:
yes, I am using the native s.port pin on r9 mini (NOT sbus for FPORT).   (don't not how to find it, can not find the inverted pad on the internet.)
internal rf was turn off.

P-I-Engineer, 2018-06-09:
that's your issue you need inverted s.port signal from your r9 mini.
i just got my 405-ctr working on sbus and s.port on softserial. I'm holding off on f.port.

MATEKSYS, 2018-06-13:
SBUS+S.Port doesn't take more Uart on F405-Wing, what is the reason to use FPort

davidngrc, 2018-06-13:
my r9 mini was broken, new one will arrived soon.
reason is simple, 1 wire less, servo only got 3 wire, and sbus+sport = 4 wire.

@MATEKSYS, could you confirm the r9 mini fport is working with the F405-Wing?
because I test at least twice, can't make it to work. 

@P-I-Engineer said it is some inverted or non-inverted thing, but I had no idea how to find a inverted or non-inverted sport in r9 mini.
could you help?

P-I-Engineer, 2018-06-13:
i don't know if there's a inverted signal or not on the mini. There is on the slim+. 

there is an issue with f.port telemetry, it will give you sensor lost alarms, it had to be throttled down to no do that.

davidngrc, 2018-07-31:
ok, I found out the FPort is working on Omnibus F4 Pro v2 + softserial on CH5 and CH6 pad + R9 mini (without the inverter) + iNav 2.0.0RC4

Wiring 
Omnibus F4 Pro CH5 pad (RX) -> 1k resister -> CH6 pad (TX)
Omnibus F4 Pro CH6 pad (TX) -> R9 mini smart port (do not need to connect the sbus wire)
CH5 CH6 pad are on https://github.com/iNavFlight/inav/issues/1346#issuecomment-335654194

in INav Configurator 2.0.0 RC3
1)  in "configuration tab", enable soft serial
2) in "port tab", enable "serial rx" on softserial1 (NO NEED to config serial rx on uart1 as we use uart1 for sbus )
3)  in "configuration tab", receiver mode set to "serial based receiver, and provider set to fport (rather than sbus)
4) in "CLI tab", set serialrx_inverted = ON
                        set serialrx_halfduplex = ON
5) now on "Receiver tab", you should able to see the channel value change while you move the x9d stick.
6) on x9d, delete all old telemetry sensor in "telemetry screen", then discover new.
7) done

davidngrc, 2018-08-01:
ok, on iNav v2.0.0RC4, after I install the sbus inverter, I still can not get FPort to work.

wiring 1):
Matek F405-WING tx2 or tx5 or sbus pin -> sbus inverter IN -> sbus inverter OUT-> R9M mini smart port.

then in iNav configurator -> set "serial rx" on port "uart2" or "uart5" or "softserial1", 
then enable FPORT in "configuration tab", 
then "CLI tab", try setting the     serialrx_inverted  and    serialrx_halfduplex   to ON or OFF.
I try different combination, no stick control, no telemetry sensor.

I even try swtich the inverter IN and OUT
wiring 2):
Matek F405-WING tx5 pin -> sbus inverter OUT -> sbus inverter IN -> R9M mini smart port.
then enable FPORT in "configuration tab", 
then "CLI tab", try setting the     serialrx_inverted  and    serialrx_halfduplex   to ON or OFF.
I try different combination, no stick control, no telemetry sensor.

davidngrc, 2018-08-01:
wiring 3):
Matek F405-WING tx2 pin -> R9M mini smart port.
then in iNav configurator -> set "serial rx" on port "softserial1"
then enable FPORT in "configuration tab"
then "CLI tab",    SET serialrx_inverted = ON    and   SET serialrx_halfduplex = ON
it work now, stick control and telemetry sensors

Here is my test dump
[inav 2.0.0RC4_MATEKF405Wing_SoftSerial_FPort_Dump.txt](https://github.com/iNavFlight/inav/files/2250191/inav.2.0.0RC4_MATEKF405Wing_SoftSerial_FPort_Dump.txt)
[inav 2.0.0RC4_MATEKF405Wing_SoftSerial_FPort_Diff.txt](https://github.com/iNavFlight/inav/files/2250192/inav.2.0.0RC4_MATEKF405Wing_SoftSerial_FPort_Diff.txt)

closed 2018-08-01 by davidngrc

terragady, 2018-08-16:
there is new R9 mini with uninverted s.port, this will work nicely with Matek f405-ctr and wing.