iNavFlight/inav #3346: Omnibus F4 V5 AIO IBUS reciever issue

gdgabq, 2018-06-09:
unable to get a response from transmitter using IBUS. Beta flight target OMNIBUSF4SD (OBSD) 3.3.1 works on same set up.
I have tried all omnibus targets in inav with similar results. All other features seem to function properly.

Flight Controller:
OMNIBUS AIO F4 V5 Flight Control
A new generation AIO Flight control for FPV, based on F405 MCU, using F405 MCU controls OSD over SPI bus in DMA mode,

povlhp, 2018-06-13:
Need more details. I am using iBus on Omnibus F4 Pro on a daily basis. 
Verify serial port settings, bind status. 
What Rx? I use x6b and iA6B.

gdgabq, 2018-06-15:
I'm using a TGY-IA6C and IA8 both bind to the transmitter. Works well with BF but when I flash INAV I don't have any response from the receiver in the configurator. All other function seem to work correctly, servos, GPS, accel...

Thanks for the assist.

closed 2018-08-28 by stale[bot]