iNavFlight/inav #3348: Stats = ON side effect

shellixyz, 2018-06-10:
When stats are enabled they are written when disarming to the EEPROM. This has the side effect of saving all the changes made to the settings with the RC adjustments like the PIDs, board alignment, etc.

digitalentity, 2018-06-10:
Indeed. How can we account for this? I mean when we write EEPROM we write it is a whole, we can't rewrite only one PG.

shellixyz, 2018-06-10:
Can't we just write the STATS_CONFIG PG alone ?

krzysztofmatula, 2018-06-12:
What about detecting an adjustment has been done and simply not saving flash in such case?

shellixyz, 2018-06-12:
@digitalentity OK right if PGs are not in different pages we can't save STATS_CONFIG alone. What about storing the STATS_CONFIG PG in a separate page then ? BTW I looked at the datasheets and iNav's code and the settings are stored in Flash there is no EEPROM in STM32Fx MCUs. Flash have 1K pages right ?

@krzysztofmatula Then the stats won't be saved unless the user saves the settings. Not a good solution.

krzysztofmatula, 2018-06-12:
@shellixyz correct, but only when a tuning is being performed, right? I don't personally use the RC adjustments. What are the typical use cases? Is it something that people have always on and ready to adjust? I'd consider it too dangerous...

shellixyz, 2018-06-12:
RC adjustments are very useful to tune the board alignment, cruise throttle, PIDs and other values in flight. I always leave them configured. It's not something used on every flight but it would be annoying to land and if not wanting to save the changes that the stats wouldn't be saved either.

shellixyz, 2018-06-12:
Also same issue when changing profile. When changing profile `setConfigProfileAndWriteEEPROM` is called and saves all the settings.

digitalentity, 2018-06-13:
@shellixyz FLASH pages are 2K on F3 devices and at least 16K on F4+ devices. Would be quite wasteful to use another page. However the whole PG storage could be redesigned in a way that allows multiple instances (revisions) of the save PG to be present in non-volatile storage. I'll think about it.

closed 2023-10-27 by DzikuVx