iNavFlight/inav #3353: No control on zero throttle with airmode enabled

silenec, 2018-06-11:
(Multirotor)
With MOTOR_STOP feature enabled and airmode (as a flight mode) enabled, I get no control at zero throttle because all motors stop. At that moment I also see ANTI_WINDUP active in stateFlags.
Is it supposed to work like this and / or how to prevent this? Do I need to disable MOTOR_STOP? I would like to have MOTOR_STOP enabled while not in airmode and disabled while airmode is active - same way it works in Betaflight.
Blackbox log is attached - time 6:12 right before a crash near the end of the log.

Thanks

[LOG00010.zip](https://github.com/iNavFlight/inav/files/2090255/LOG00010.zip)

shellixyz, 2018-06-12:
Will be fixed in 2.0 (#3298)

silenec, 2018-06-12:
Are you sure? To fix this, I would need to keep the motors running even at 0% throttle with motor_stop and airmode enabled. Not sure if #3298 does that?

silenec, 2018-06-12:
In other words, I need airmode to override motor_stop. Just like in Betaflight: https://github.com/betaflight/betaflight/issues/634

shellixyz, 2018-06-12:
It does not only do that but it does. It will stop the motor if the throttle is bellow `mincheck` only for fixed wing and for MRs only if air mode is not active.

silenec, 2018-06-12:
Oh ok, then it's all good. Thanks!

silenec, 2018-06-19:
So far #3298 seems to be working, at least on the table. Motors are now stopped without airmode and spinning with airmode activated. We can close this issue.

closed 2018-06-19 by silenec