iNavFlight/inav #3355: Mixers dedicated for aircraft

Artyum, 2018-06-12:
Hi, I've successfuly fly with iNav on Multiplex Mentor with F4v3 board. The stabilization, RTH and OSD are fantastic :) The only thing I really miss are mixers dedicated for airplane like differential airleons or airleon to rudder. Without this the plane is flying in turns more like a brick and it is far from being smooth. Is it possible to setup those mixers?

DzikuVx, 2018-06-12:
INAV 2.0 will have a GUI in Configurator to setup mixers in an easy way. This will look more less like this: https://www.youtube.com/watch?v=0cLFu-5syi0

closed 2018-06-12 by DzikuVx

Artyum, 2018-06-12:
New GUI for mixers will be great feature but I also have no problem using CLI just ask if above mixers can be set in current v1.9 :)

DzikuVx, 2018-06-12:
Then you have to use CUSTOM_AIRPLANE mixer and setup everything like it's
described in docs