iNavFlight/inav #3356: Launch Mode improvements

craig34a, 2018-06-12:
1. Allow full power before launch. This makes it easier to get the plane to flying speed. 
2. Add climb to pre-set altitude, say 50 or 100 m to be determined by setup. 

The Parrot Disco has this feature and it makes it extremely easy and safe to launch. The propeller spins before hand launch to give full thrust from the moment it flies. Once airborne it climbs to a pre-set altitude and enters a Position Hold pattern. 

### If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

### Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://www.rcgroups.com/forums/showthread.php?t=2495732), [Slack](https://inavflight.signup.team) or other user support forums & groups (e.g. facebook).

### Board and Version
_Board name goes here. If you have a link where you've purchased the board - please include it_
_Version of INAV used goes here_

_Use CLI command `version` and put its output here_

_If it's a custom-compiled firmware please mention this_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
* _Expected Results_
* _Actual Results_

_Is this a new feature request?_

#### Blackbox logs/Config (.ZIP) where problem occurs
_Go to CLI, execute `dump` command copy its output to [PasteBin](https://pastebin.com) and provide a link to a paste here.

_Upload a zipped blackbox log (if available)_

giacomo892, 2018-06-12:
Hi. You can already do the same with INAV.
Set `nav_fw_launch_idle_thr` to the desidered value. As soon you will rise the throttle motor will start spinning. 
You can then set the altitude at which aborting the launch and switch to position hold by setting `nav_fw_launch_max_altitude`and `nav_fw_launch_timeout` long enough to reach that altitude, and just enable position hold right after you enabled launch mode so it will enter position hold after the launch.

shellixyz, 2018-06-12:
With launch mode enabled you can also enable RTH and the plane will circle above home at the set RTH altitude once it has been launched for at least `nav_fw_launch_timeout` or after reaching the `nav_fw_launch_max_altitude` altitude.

craig34a, 2018-06-12:
That's great, thank you. If we can add this setting to configurator set-up screens that would make it more accessible.