iNavFlight/inav #3358: Feature Request: Plane/Wing slow landing

davidngrc, 2018-06-12:
The fixed wing or flying wing need a lot of ground to land smoothly.
Could we make the nose go up, and the tail go down, and slowly move toward to the Home point?

For example if you are flying in a beach, with above method, the plane can just nose up in the sea within home point like 20 meters, then slowly toward home point, this way, we don't need the entire beach as a landing strip to land a plane.

like in this video https://www.youtube.com/watch?v=Yz1hZF_bhfE&t=3m10s
from 3:10 to 3:20

DzikuVx, 2018-06-12:
Currently this is rather a no-go. We have no means to control airspeed to prevent stalls.

digitalentity, 2018-06-13:
It's not about falling. @DzikuVx is talking about stalling - when place speed relative to the air is not enough to control it. Flaring is the most tricky part of landing - you still have to maintain the lowest possible airspeed to keep machine controllable.

davidngrc, 2018-06-13:
How about something like this?
this method does not rely on air speed.

aim_z = 100cm // 1 meter or home_z
plane change to verticle;
throttle change to max throttle; // like 2000, now the plane should go vertical up and not stall.

while( gps_z != aim_z  ||  gps_x != home_x  ||  gps_y != home_y )
{
	if( gps_z > aim_z )
	{
		throttle -= 5; // decrease throttle,  1000 throttle (no throttle) / 5 = 200 cycle
		sleep( 100ms ); // 200*100ms = 20000 ms = 20 seconds ( from max throttle to no throttle need 20 seconds)
	}
	else
	{
		throttle ++; // increase throttle
		sleep( 10ms );
	}

	move flap toward home x and y;
}

closed 2018-08-26 by stale[bot]