iNavFlight/inav #3359: Delayed / averaging sensors causing strange behaviour - Flying wrong direction

povlhp, 2018-06-12:
I have a video here clearly showing delayed OSD output - or more likely 10s averaging.
https://youtu.be/j-Gj1-ygebg

Bix3, Omnibus F4 Pro, built-in OSD, iNav 1.9.1. No compass.

But it seems like these delayed values are also used for navigation, which can be pretty bad. I have my plane coming home in RTH mode. Turned on 250 meters away after flying like 45  degrees right of target for 10s. Almost straight line. Home pointer is fine.

Plane turns left, too much to the left as the averaging makes it overshoot the direction. Flies to like 160m left of home. Now it finds out it is too much to the left, so it keeps banking right, and does a narrow 270 degrees turn, keeping the home arrow pointing right, and at some point it starts moving all the way to the left. Ending up flying away from home. I get scared and starts giving throttle and correct the flight.
OSD shows no airspeed, but with delay it shows the real 70+ km/h = 20+ m/s. When I get the plane turned around and pointing home, the home distance still increases, and the pointer is still the wrong way, until I have a stable heading for 10s.

OSD shows the issue on this video, RTH completely misjudging direction. First 3-4 secs are same image due to my cut on a non-keyframe
https://youtu.be/rObMzWoJC8o

Can't find an obvious setting to change averaging period.

I imagine a compass will help. I can also imagine RTH keep overshooting and never retun to home.

digitalentity, 2018-06-13:
Config dump and blackbox? Looks like a tuning/setting issue to me. OSD data is not delayed, airplane response to direction change is - usually the case when airplane don't have enough control authority to turn fast enough in automatic modes or navigation PIDs are messed up.,

povlhp, 2018-06-13:
If you look at the first video, you see delayed OSD. 10s from impact until OSD stops counting down. 2nd video clearly shows the same.

I think the issue is the late update of data, maybe something in the M8N clone or communications with it.

Here is screenshot of the path, it clearly has authoritative control. The 270-ish turn is registered as using a single car lane only. That is 10m diameter. So plenty of authority (as can bee seen from 2nd video)
[image]

Blackbox here.  Config not available right now. Will come later when I am home (@1800 CEST)
[LOG00008.TXT](https://github.com/iNavFlight/inav/files/2096942/LOG00008.TXT)

Again, I think it has something to do with either filtering/smoothing of GPS data, or a delay/buffering somewhere. It is very clear that the OSD is like 10s out of sync (delayed) compared to actual plane direction/movements. Thus video is important.

I opened another one with less data as #3316, just close this as duplicate. I no longer think the OSD is delayed, I think OSD is displaying what the FC thinks is correct data. But the FC i using old stale data, or some heavily postprocessed data.

2 people on Facebook seems to have experienced strange home pointer direction as well on RTH. Might be related.

povlhp, 2018-06-13:
Another comment, when launching, OSD speed  is not moving above 0 m/s for the first 7 seconds or so.  Same goes for DST to home.
It launches/flies stabilized,but that is onboard gyro, not GPS. So again, that helps confirm my theory that it is navigational data from GPS that is somehow delayed, buffered, cached, filtered, averaged or whatever. Plane flies nice when I fly in horizon, angle or manual. Only GPS navigation untrustworthy.

digitalentity, 2018-06-13:
Replace the GPS module.

povlhp, 2018-06-13:
I accept that replacing the GPS would be natural in troubleshooting. Will do that. I wonder why 2 persons more had the same issue. Will ask them on facebook, maybe we can narrow it down to the HGLRC mini GPS module

povlhp, 2018-06-13:
One report on Facebook is using FuriousFPV GPS module, same as HGLRC. Other RTFQ M8N (not sure of size). Now that we know what to look for, we will all look for delayed OSD, and repeatability hereof. Might not happen all the time.  So all with chinese M8N clones.

Plans is to test / evaluate my module before replacing it. Both with OSD, and with uCenter.
See if it is consistent. I have my logic analyzer I could use as well to grab the traffic as a last step. Think it is important to localize the issue, and warn.

povlhp, 2018-06-13:
Could not provoke the error today. Tried many times in garden walking around. Home pointer always correct (if turned on facing north). Walk/run and stop, and speed was 0 within 3 secs.

Here is my complete config:
`# dump

# version
# INAV/OMNIBUSF4PRO 1.9.1 Apr 21 2018 / 13:12:44 (03a5c1922)

# resources

# mixer
mixer AIRPLANE

mmix reset

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1575 100 -1 
servo 3 1000 2000 1469 -100 -1 
servo 4 1000 2000 1469 -100 -1 
servo 5 1000 2000 1597 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# feature
feature -RX_PPM
feature -VBAT
feature -TX_PROF_SEL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature -TRACE
feature VBAT
feature TX_PROF_SEL
feature MOTOR_STOP
feature GPS
feature BLACKBOX
feature AIRMODE
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
[... 504 lines cut ...]
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_mc_heading_p = 60
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set nav_fw_heading_p = 60
set thr_mid = 50
set thr_expo = 0
set tpa_rate = 0
set tpa_breakpoint = 1500
set fw_tpa_time_constant = 0
set rc_expo = 70
set rc_yaw_expo = 20
set roll_rate = 20
set pitch_rate = 15
set yaw_rate = 9
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100`

mrduncan626, 2018-06-13:
Since you are not using a compass, then iNav has to calculate a direction vector based on individual GPS data points.  So, depending on your GPS data rate and the number of data points iNav requires to calculate a direction vector could be the cause of the OSD data delay.  I thought most GPS default data rate was 5Hz.  However, I would think GPS data rates set to 1Hz or greater could cause OSD data delays.

povlhp, 2018-06-13:
I understand the GPS data must be used to calculate say a vector. This would need at least 2 data points, preferable far enough from each other to sufficiently overcome drift-like errors.
The problem here is different, likely caused be some bad state in the M8N clones. We are now at least 4 persons on FB that have experienced homepointer is very bad (typically seen as opposite direction).

I am the only one of the 4 that has OSD / DVR video. And on the video, it is clear that we have a very slow update rate. Lets say 10s delay. Seems pretty constant that speed, homepointer etc is displayed as perfect data for where the quad was exactly 10s ago. Even lattitude/longtitude goes up/down close to 10s off.

Since we do not buffer data that long on the F4, it is only a few bytes, and thus few messages. And they will be pulled from the main loop faster than they arrive.

So the conclusion for now, is that the issue is with the GPS modules hitting some strange state.

closed 2018-08-26 by stale[bot]