iNavFlight/inav #3378: [feature request] specific failsave directive in waypoint mode (mission)

matoy, 2018-06-16:
Hi,

That would be nice to be able to select a specific different failsafe directive for waypoint mission.

For example, in general, failsafe would trigger a defined action, except in waypoint mode where it would trigger something else, let's say do nothing (or RTH).

That would be useful for situations where you send the drone to a mission where you might loose RX link but you want it to make it anyway and get link back when it comes back. If you fly over a forest, that would avoid the quad to land on top of trees.

In general, i would not fly without a defined failsafe action but in waypoint mode, that could be nice (if you trust your build).

Thank you.

closed 2018-08-29 by stale[bot]

access3424, 2018-11-13:
Yes! lets do that!

shellixyz, 2018-11-13:
Duplicate of #770, #977, #2340 and #2972

matoy, 2018-11-18:
Hi,

Since this is multiple duplicate; any chance to get this feature any time soon?
Thanks.

shellixyz, 2018-11-20:
@matoy Honestly can't say. iNav is a collaborative project it will be done if somebody is interested and has time to work on it. Personally I'm working on other things right now and won't have time to do this soon. Plus I can hardly do any testing right now because of the winter weather. Contributions are always welcome.