iNavFlight/inav #3384: Omnibus F4 nano v6

craigywsm, 2018-06-17:
### If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

### Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://www.rcgroups.com/forums/showthread.php?t=2495732), [Slack](https://inavflight.signup.team) or other user support forums & groups (e.g. facebook).

### Board and Version
_Board name goes here. If you have a link where you've purchased the board - please include it_
_Version of INAV used goes here_

_Use CLI command `version` and put its output here_

_If it's a custom-compiled firmware please mention this_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
* _Expected Results_
* _Actual Results_

_Is this a new feature request?_

#### Blackbox logs/Config (.ZIP) where problem occurs
_Go to CLI, execute `dump` command copy its output to [PasteBin](https://pastebin.com) and provide a link to a paste here.

_Upload a zipped blackbox log (if available)_

craigywsm, 2018-06-17:
Found answer at 
https://github.com/iNavFlight/inav/pull/3294

craigywsm, 2018-06-21:
Not able to communicate with board with the latest build. Help please

fast12learn, 2018-08-20:
Sorry for the betaflight dump. I'm flashing iNav 2.0 RC6 via configurator 2.0 RC3. The only thing I change in iNav is to set half duplex "on" and I check the appropriate iNav configurator boxes for Spektrum DSMX (2048) satellite. There are no hardware changes between the betaflight and iNav. The craft flies on betaflight. iNav currently does not receive the dsmx signal.

stronnag, 2018-08-20:
Please post an iNav diff. DSMX works perfectly on iNav.

fast12learn, 2018-08-21:
I'm sorry I believe that I was too vague and maybe offensive.

DSMX has worked perfectly on every other iNav target I've tried except this one. I have two matek f405 wings with the exact same DSMX configuration. I've even used it on f3 Omnibus and f1 units. I'm very interested in quads and wings that weigh less than 250 grams. This would be a great board for that.  Airbot support states that I should use the Firework V2 target in iNav the Dsmx signal should go to UART 

Is this correct or should I use a different target?

I'll reflash to inav 2.0-RC6 Fireworks v2 target and then provide a dump as soon as I can

fast12learn, 2018-08-23:
This is the "diff" stable release candidate for inav 2.0 stable with the fireworks v2.0 target for the Airbot F4 nano v6. I have a light that says I am connected with my transmitter. I have no indication that the signal is being processed by the board. When I flash to Betaflight the only command I give on the cli is set rx half duplex to on and everything works. I wand iNav not beta flight. Am I using the wrong target? Have I missed something in the configuration?

# diff

# version
# INAV/FIREWORKSV2 2.0.0 Aug 19 2018 / 21:25:18 (dbdd1656a)

# resources

# mixer
mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix

# servo

# feature
feature MOTOR_STOP
feature PWM_OUTPUT_ENABLE

# beeper

# map
map TAER

# serial

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# osd_layout

# master
set looptime = 500
set align_gyro = CW90
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 90
set gyro_notch1_hz = 400
set gyro_notch1_cutoff = 300
set gyro_notch2_hz = 200
set gyro_notch2_cutoff = 100
set align_acc = CW90
set acc_hardware = MPU6000
set acczero_x = 144
set acczero_y = -3
set acczero_z = 39
set accgain_x = 4099
set accgain_y = 4089
set accgain_z = 4046
set mag_hardware = NONE
set baro_hardware = BMP280
set pitot_hardware = NONE
set serialrx_provider = SPEK2048
set serialrx_halfduplex = ON
set motor_pwm_rate = 2000
set motor_pwm_protocol = ONESHOT125
set model_preview_type = 3
set attitude_task_frequency = 100

# profile
profile 1

set mc_p_pitch = 48
set mc_i_pitch = 50
set mc_d_pitch = 22
set mc_p_roll = 36
set mc_i_roll = 40
set mc_d_roll = 20
set mc_p_yaw = 70
set dterm_lpf_hz = 90
set dterm_notch_hz = 260
set dterm_notch_cutoff = 160
set rc_yaw_expo = 70
set roll_rate = 75
set pitch_rate = 75
set yaw_rate = 75

# battery_profile
battery_profile 1

fast12learn, 2018-08-24:
Sorry! Found It!!! The default in the configurator is to enable soft serial. You must disable soft serial as it conflicts with UART 1/dsm serial receivers.

closed 2018-11-22 by stale[bot]

jasc76, 2018-11-26:
Had a successful first inav flight on an airbot v6 nano. Great work on a little board ;-)
One thing that is missing is a current sensor. I cant use ESC Tel, so I need to rely on an external shunt for measurement (hglrc xt30 with current sensor output)
Well, the nano has no current sensor input. On the ESC connector you only have RX4 (PA1). On Betaflight some folks used resource remapping, which of course is not available here. 
So is there a chance to have a current sensor on PA1 or another solution ?

mozgoverflow, 2019-05-13:
@jasc76
How you set up compass on it? I don't see I2C pads (SDA, SCL) on Omnibus F4 Nano V6.

Elaugaste, 2020-01-20:
> Had a successful first inav flight on an airbot v6 nano. Great work on a little board ;-)
> One thing that is missing is a current sensor. I cant use ESC Tel, so I need to rely on an external shunt [...]

I find inav with current via rx4 
https://drive.google.com/drive/folders/1js_i67zCl0fkSaZm3Tbxlb1D2l2XHPyq