iNavFlight/inav #3391: [GPS/POSHOLD] Problem with poshold, rth, slow drift in a random direction

ValCx, 2018-06-18:
Hi everyone,

My quad was working very well with inav 1.7.2 : rth, poshold, waypoint. During winter I let him away due to bad weather. 

Since I get it back I have some problem with Gps Mode : Poshold and RTH,
when I activate poshold, nothing append, the quad drift away. For RTH mode, the quad turn to the home and go in a wrong direction. If I adjust to go to the home, the quad detect home and land.

I tried to re-calibrate compass, accelerometer. I tried with older 1.7.2 version. I tried with internal compass. I tried with another GPS. I tried to re-align compass. Nothing affects, I already have the problem.

I attach a logfile and a dump of my configuration. The logfile show a straight flight, some poshold and rth mode.
If someone has an Idea he is welcome!

Thanks

Valentin

### Board and Version
Openpilote Revolution
FW 1.9.1
Gps/compass neo-m8n and pam-7q for test

#### DUMP
[blackbox_log_2018-06-18_200047.TXT](https://github.com/iNavFlight/inav/files/2112410/blackbox_log_2018-06-18_200047.TXT)

Entering CLI Mode, type 'exit' to return, or 'help'

# dump

# version
# INAV/REVO 1.7.3 Sep  9 2017 / 10:40:06 (974440df)

# resources

# mixer
mixer QUADX

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
[... 448 lines cut ...]
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

#

Risineo, 2018-06-19:
Sounds like you did not have ANGLE or HORIZON mode turned on when switching to RTH or POSHOLD. Both RTH and POSHOLD require you to also have ANGLE or HORIZON active at the same time.

Mick51, 2018-06-20:
I have exactly the same problem, whether with Angle activate or horizon, there is a bug in Inav 1.9.1

stronnag, 2018-06-20:
@ValCx  you have a pre-1.9 setting for`nav_mc_vel_xy_p`; the release note explains that the scaling has changed. A value in the range 35-40 should be more appropriate.  Please see the release note and retest with a proper value.

Note: If you are restoring settings between versions, it is advisable to read the release notes, and remove any modified variables, so the defaults will be restored.

Please let us know if this resolves your issue. If you still have a problem, a blackbox log will be necessary for further analysis.

stronnag, 2018-06-20:
Additionally, please also confirm that the baro is well covered with open cell foam in order to protect it from sunlight and wind. The baro is behaving somewhat erratically; you may also want it reset on each arm rather than first arm.

The GPS bahaviour is marginal for navigation modes, it may be that you have other emitters (VTX for example) causing intererence that degrades GPS performance; for a M8N one would expect at least 12 satellites. HDOP > 2 I would consider marginal and would not fly under those conditions.

These, together with the obsolete `nav_mc_vel_xy_p` setting may explain the behaviour. 

The machine is not flying away on RTH, it is attempting to gain altitude  prior to turning for home. The combination of poor baro performance, marginal GPS and inappropriate `nav_mc_vel_xy_p` may contribute to the long time (5+seconds) in this phase.

The wandering route home is typical of inappropriate vel_xy_p and marginal GPS behaviour. It is noticeable how the craft flys directly in the home direction once the HDOP drops below 2.

Summary. Fix your settings, sort out your sensors. Some basic tuning is required.

zipray, 2018-06-25:
My FC is F3 evo. After inav 1.9, I found that the default compass direction of MPU9250 is not correct, and it must be set to flip90CW.I adjusted nav_mc_vel_xy_p=35, calibrated the compass and switched to return mode in the case of GPS hop<2, and the aircraft would still start to wander back.After entering loiter mode, the flight trajectory is that the spiral will fly farther and farther, and then enter ANGLE mode automatically when it exceeds the maximum return range.

giacomo892, 2018-06-25:
@zipray are you sure have you selected to use external compass instead of the internal one?

ValCx, 2018-06-25:
@stronnag
Thanks for answers, here it is a dump with the old version 1.7.3, with all params reset and re-configured.
I updated to 1.9.1 with a reset of all params, I will let a dump and a new log with this version soon.
mag_to_use is set to 1, to select the external compass.

I will try with nav_mc_vel_xy_p=35 and min sat=12. 
The Althold mode is ok, and Angle mode is activate during poshold.

teckel12, 2018-06-25:
@ValCx I've noticed this type of behavior on rare occasion because I really don't have a good GPS lock.  What I notice is that even though INAV says the HDOP is good and I have 6+ satellite locks, the GPS altitude is wrong.  I know where I launch is about 600' above sea level.  But, I've seen it be hundreds of feet off (one time -2700').  When this happens, I get a lot of drift until it finally gets a good GPS lock.

In any case, monitor GPS altitude and verify with your phone they basically match and it's not moving all over the place (classic sign of a bad GPS lock).

ValCx, 2018-07-13:
@teckel12 I tried with a new gps cam m8. I get  10+ sats, the gps lock is pretty good HDOP = 1.6, it's not moving all over the place. And I still have the problem.

teckel12, 2018-07-13:
Are you sure `set align_mag = CW270FLIP` is correct?

ValCx, 2018-07-17:
I'm pretty sure, I align my quad with a compass ans he is pretty good for each point : 0 90 180 and 270.
Here it is a dump of my last configuration with inav 1.9.1 : 
detected

Entering CLI Mode, type 'exit' to return, or 'help'

# dump

# version
# INAV/REVO 1.9.1 Apr 21 2018 / 13:17:09 (03a5c1922)

# resources

# mixer
mixer QUADX

mmix reset

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# feature
feature -RX_PPM
feature -VBAT
feature -TX_PROF_SEL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature -TRACE
feature VBAT
feature GPS
feature PWM_OUTPUT_ENABLE

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
[... 454 lines cut ...]
set nav_mc_heading_p = 60
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set nav_fw_heading_p = 60
set thr_mid = 50
set thr_expo = 0
set tpa_rate = 0
set tpa_breakpoint = 1500
set fw_tpa_time_constant = 0
set rc_expo = 70
set rc_yaw_expo = 20
set roll_rate = 50
set pitch_rate = 50
set yaw_rate = 45
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100

#

closed 2018-09-29 by stale[bot]