iNavFlight/inav #3398: Problems with magnetometer since using INAV

elektrofreakz, 2018-06-19:
### Board and Version
Omnibus F4 V3 from ebay seller
Inav 1.9.3

# INAV/OMNIBUSF4V3 1.9.1 Apr 21 2018 / 13:13:58 (03a5c1922)

### Behavior
When using the heading hold mode on inav the quad starts spinning around. I never had this problem with this magnetometer in cleanflight, it has started since using INAV
GPS hold and altitude hold work fine. I use them together to get my drone as stable as possible for photography. When i undo the heading hold function it stops spinning.

I checked my magnetometer function by looking in the sensors tab from INAV and spinning the motors. It does not cause any interference. Magnetometer is a build in from an Ublox M8N module. And is mounted +- 10Cm above quad. Frame is a 3D printed MHQ2.

Also did multiple calibrations but did not help. Only going back to cleanflight solves this?

#### Blackbox logs/Config (.ZIP) where problem occurs
pastebin CLI dump:
https://pastebin.com/hukweiaD

rb1205, 2018-06-22:
Not sure if it's related with this, however I've got magnetometer issues since 1.9 as well.

elektrofreakz, 2018-06-22:
According to my compass chip in the M8N it's a L8832122 AKA HMC5883L. So cant imagine thats the problem, but i will try downgrading to 1.8 and test my setup!

[image]

elektrofreakz, 2018-06-22:
tried 1.8 and lower, but i'm not allowed anymore toe recalibrate my compass?
[image]

In the magnetometer selection tab i can only use the MC5883 setting. Other ones are not recognized by the omnibus. So, according to that i am sure i have a MC5883 chip

elektrofreakz, 2018-06-22:
still turning around with INAV1.8. Can't calibrate the compass with the calibrate button. INAV recognizes the compass as a HMC5883. I tried manually setting the correct value's (from previous dump) via CLI commands. After doing this the red cross behind the compass line in the setup tab went away,. But it is still turning around. It's more steady with GPS and compass off. The calibrate problem was the same with the accelerometer. Had to set the variables in CLI manually

rb1205, 2018-06-23:
The GUI calibrate buttons are a recent addition, might not work on 1.8. Use the [stick command](https://discourse-cdn-sjc2.com/standard16/uploads/micro_motor_warehouse/optimized/2X/2/2e73a262e72541e41c2a3ae9fac1bdb5ef94fcaa_1_690x302.png) to trigger it.

elektrofreakz, 2018-06-25:
Did the stick command and saved settings, will try flying the craft tomorrow. Also noticed now that when i tilt the craft forward, the model in the setup screen turns around. When i check the INAV docs it says that i should recalibrate my compass? 

> If heading value is correct when copter is level but drifts when you tilt the model, then your should re-calibrate the compass.

Did this multiple times, but keeps drifting when tilted. Can this be the strange turning around that i am experiencing?

shellixyz, 2018-06-25:
@elektrofreakz It probably means your magnetometer alignment setting is not correct (the magnetometer is upside down in the GPS module).

fiam, 2018-06-25:
@elektrofreakz Don't fly until you have that sorted out. Enabling any navigation modes with an improperly setup compass it's going to end up badly.

elektrofreakz, 2018-06-25:
Yeah, i know lol. Had to rebuild the whole craft because i tried to fly with not enough GPS satellites and returning home mode.

Got it now on CW 180 FLIP. It looks like this should be the right setting. The heading is also reading almost 0 degrees when the craft nose is pointing north. It doesnt react as extreme as before. But still when i tilt the craft forward it turns around 20 degrees to the right. When i tilt it backwards its does the same but to the left

elektrofreakz, 2018-06-26:
Had a small testflight, and it seems to be stable. Had no issues with turning in flight.

teckel12, 2018-06-26:
@elektrofreakz You stated "But still when i tilt the craft forward it turns around 20 degrees to the right. When i tilt it backwards its does the same but to the left". This is a classic indication that you don't have the correct orientation for your compass. It may fly, but in other situations it could be catastrophic. I can all but guarantee that your compass orientation is incorrect.

elektrofreakz, 2018-06-27:
Yes, i tried all different MAG positions, and recalibrate the MAG sensor. But with CW 180deg it's the most stable. Cant say its completely stable but it's better then nothing. Only issue is that with the 180deg flip the heading is 180 deg off. So north is not facing north..

teckel12, 2018-06-28:
@elektrofreakz Sounds like a magnatometer alignment problem still, try `set align_mag = CW270FLIP` and then calibrate outside.

elektrofreakz, 2018-07-02:
Fixed! Got an other GPS and magnetometer. Got the SE100 now.
got the magnetometer on CW270 flip.
Tried the old one on those settings but could not get it right. Also, the GPS on the SE100 is better :)

https://youtu.be/S1Hf1dx2XvY

closed 2018-09-14 by stale[bot]