iNavFlight/inav #3402: Altitude estimate is incorrect despite accurate GPS fix and barometer

aubreyrjones, 2018-06-20:
### Board and Version

Matek F405-WING
INAV 2.0.0 - `development` branch
INAV/MATEKF405SE 2.0.0 Jun 19 2018 / 18:46:07 (d57136b)

This is my built from my working fork at https://github.com/aubreyrjones/inav/tree/osd_custom_string . However, testing by others in the INAV Telegram chat indicates that the mainline `development` branch should show identical behavior.

### Behavior

My craft is configured with UBLOX GPS and on-board barometer.

When I boot up the flight controller on the bench using only USB power, so that the barometer is powered but the GPS is not powered, the estimated altitude of the craft is displayed as 0m for a couple of seconds before it dips downward to a couple meters then upward to hundreds of meters. It sometimes oscillates several times with positive gain before settling at a ridiculous value. My latest test gives the altitude of my unmoved, benched aircraft as being 212m. This is displayed both in the Sensors tab of the configurator and as an altitude of 0.6Kft in the OSD.

If I boot the same board from main flight battery with warm GPS and good 3D fix with correct elevation above sea level, again the altitude is displayed as 0 for a moment... but then rises to several dozen meters, before dropping to -500m. This is displayed both on OSD as -1.7Kft and also in the Sensors tab. The altitude displayed in the GPS tab is only 1505m.

[image]

[image]

giacomo892, 2018-06-20:
Hello.
Have you messed with `inav_w_z_baro_p`setting? Can you provide a `diff`?
Are you using `FIXED WING` or `MULTIROTOR`?

aubreyrjones, 2018-06-20:
I'm using fixed wing.
```
# diff

# version
# INAV/MATEKF405SE 2.0.0 Jun 19 2018 / 19:03:38 (d57136b)

# resources

# mixer
mmix 0  1.000  0.000  0.000  0.000

# servo mix
smix 0 3 0 50 0
smix 1 3 1 50 0
smix 2 4 0 -50 0
smix 3 4 1 50 0

# servo
servo 3 1100 1765 1459 -100 
servo 4 1100 1732 1400 100 

# feature
feature -SOFTSERIAL
feature MOTOR_STOP
feature GPS
feature PWM_OUTPUT_ENABLE

# beeper

# map

# name
name Corina

# serial
serial 3 2 115200 38400 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1925 2100
aux 1 10 1 1900 2100

# adjrange

# rxrange
rxrange 0 988 2011
rxrange 1 988 2011
rxrange 2 988 2011
rxrange 3 988 2011

# osd_layout
osd_layout 0 0 26 11 V
osd_layout 0 1 24 0 V
osd_layout 0 3 8 6 H
osd_layout 0 4 8 6 H
osd_layout 0 7 0 11 V
osd_layout 0 9 1 2 H
osd_layout 0 11 20 0 V
osd_layout 0 12 24 1 V
osd_layout 0 13 5 0 V
osd_layout 0 14 0 11 H
osd_layout 0 15 0 0 V
osd_layout 0 20 17 12 V
osd_layout 0 21 2 12 V
osd_layout 0 24 13 0 V
osd_layout 0 28 23 11 H
osd_layout 0 30 1 10 V
osd_layout 0 49 23 0 H
osd_layout 0 50 12 0 H
osd_layout 0 51 20 11 V
osd_layout 1 0 8 6 H
osd_layout 1 1 8 6 H
osd_layout 1 2 23 8 H
osd_layout 1 3 23 9 H
osd_layout 1 4 12 12 H
osd_layout 1 5 20 2 H
osd_layout 1 6 1 2 H
osd_layout 1 7 8 6 H
osd_layout 1 8 1 3 H
osd_layout 1 9 1 4 H
osd_layout 1 10 23 1 H
osd_layout 1 11 0 11 H
osd_layout 1 12 1 0 H
osd_layout 1 13 2 10 H
osd_layout 1 14 2 11 H
osd_layout 1 15 2 12 H
osd_layout 1 16 15 1 H
osd_layout 1 17 18 12 H
osd_layout 1 18 0 12 H
osd_layout 1 19 14 11 H
osd_layout 1 20 1 1 H
osd_layout 1 21 12 2 H
osd_layout 1 22 23 5 H
osd_layout 1 23 24 7 H
osd_layout 1 24 3 5 H
osd_layout 1 25 23 11 H
osd_layout 1 26 23 12 H
osd_layout 1 27 1 13 H
osd_layout 1 28 0 10 H
osd_layout 1 29 12 1 H
osd_layout 1 30 6 2 H
osd_layout 1 31 18 2 H
osd_layout 1 32 1 5 H
osd_layout 1 33 1 5 H
osd_layout 1 34 1 6 H
osd_layout 1 35 1 7 H
osd_layout 1 37 1 2 H
osd_layout 1 38 1 8 H
osd_layout 1 39 1 7 H
osd_layout 1 40 0 0 H
osd_layout 1 41 0 0 H
osd_layout 1 42 0 0 H
osd_layout 1 44 3 6 H
osd_layout 1 45 3 7 H
osd_layout 1 46 23 0 H
osd_layout 1 47 12 0 H
osd_layout 1 48 0 0 H
osd_layout 1 49 8 6 H
osd_layout 1 50 8 6 H
osd_layout 1 51 23 8 H
osd_layout 2 0 23 9 H
osd_layout 2 1 12 12 H
osd_layout 2 2 20 2 H
osd_layout 2 3 1 2 H
osd_layout 2 5 1 3 H
osd_layout 2 6 1 4 H
osd_layout 2 7 23 1 H
osd_layout 2 8 0 11 H
osd_layout 2 9 1 0 H
osd_layout 2 10 2 10 H
osd_layout 2 11 2 11 H
osd_layout 2 12 2 12 H
osd_layout 2 13 15 1 H
osd_layout 2 14 18 12 H
osd_layout 2 15 0 12 H
osd_layout 2 16 14 11 H
osd_layout 2 17 1 1 H
osd_layout 2 18 12 2 H
osd_layout 2 19 23 5 H
osd_layout 2 20 24 7 H
osd_layout 2 21 3 5 H
osd_layout 2 22 23 11 H
osd_layout 2 23 23 12 H
osd_layout 2 24 1 13 H
osd_layout 2 25 0 10 H
osd_layout 2 26 12 1 H
osd_layout 2 27 6 2 H
osd_layout 2 28 18 2 H
osd_layout 2 29 1 5 H
osd_layout 2 30 1 5 H
osd_layout 2 31 1 6 H
osd_layout 2 32 1 7 H
osd_layout 2 33 1 5 H
osd_layout 2 34 1 2 H
osd_layout 2 35 1 8 H
osd_layout 2 36 1 7 H
osd_layout 2 37 0 0 H
osd_layout 2 38 0 0 H
osd_layout 2 39 0 0 H
osd_layout 2 40 0 0 H
osd_layout 2 41 3 6 H
osd_layout 2 42 3 7 H
osd_layout 2 43 23 0 H
osd_layout 2 44 12 0 H
osd_layout 2 46 8 6 H
osd_layout 2 47 8 6 H
osd_layout 2 48 23 8 H
osd_layout 2 49 23 9 H
osd_layout 2 50 12 12 H
osd_layout 2 51 20 2 H
osd_layout 3 0 1 2 H
osd_layout 3 1 8 6 H
osd_layout 3 2 1 3 H
osd_layout 3 3 1 4 H
osd_layout 3 4 23 1 H
osd_layout 3 5 0 11 H
osd_layout 3 6 1 0 H
osd_layout 3 7 2 10 H
osd_layout 3 8 2 11 H
osd_layout 3 9 2 12 H
osd_layout 3 10 15 1 H
osd_layout 3 11 18 12 H
osd_layout 3 12 0 12 H
osd_layout 3 13 14 11 H
osd_layout 3 14 1 1 H
osd_layout 3 15 12 2 H
osd_layout 3 16 23 5 H
osd_layout 3 17 24 7 H
osd_layout 3 18 3 5 H
osd_layout 3 19 23 11 H
osd_layout 3 20 23 12 H
osd_layout 3 21 1 13 H
osd_layout 3 22 0 10 H
osd_layout 3 23 12 1 H
osd_layout 3 24 6 2 H
osd_layout 3 25 18 2 H
osd_layout 3 26 1 5 H
osd_layout 3 27 1 5 H
osd_layout 3 28 1 6 H
osd_layout 3 29 1 7 H
osd_layout 3 30 1 5 H
osd_layout 3 31 1 2 H
osd_layout 3 32 1 8 H
osd_layout 3 33 1 7 H
osd_layout 3 34 0 0 H
osd_layout 3 35 0 0 H
osd_layout 3 36 0 0 H
osd_layout 3 37 0 0 H
osd_layout 3 38 3 6 H
osd_layout 3 39 3 7 H
osd_layout 3 40 8 1 H
osd_layout 3 41 8 28 H
osd_layout 3 42 4 3 H
osd_layout 3 43 18 3 H
osd_layout 3 44 15 27 V
osd_layout 3 46 1 0 H
osd_layout 3 47 0 0 H
osd_layout 3 48 0 0 H
osd_layout 3 49 0 0 H
osd_layout 3 50 0 0 H
osd_layout 3 51 10 10 V
osd_custom_string KE0SDQ

# master
set acc_hardware = MPU6000
set acczero_x = 70
set acczero_y = -17
set acczero_z = -220
set accgain_x = 4097
set accgain_y = 4075
set accgain_z = 4008
set mag_hardware = NONE
set baro_hardware = BMP280
set pitot_hardware = NONE
set rssi_channel = 8
set serialrx_provider = CRSF
set min_throttle = 1215
set max_throttle = 2000
set motor_pwm_rate = 50
set model_preview_type = 8
set small_angle = 180
set inav_w_xy_flow_p =  0.500
set inav_w_xy_flow_v =  0.500
set inav_w_z_gps_v =  0.500
set inav_w_z_res_v =  0.010
set inav_w_xy_res_v =  1000.000
set inav_w_acc_bias =  100.000
set osd_row_shiftdown = 1
set osd_units = IMPERIAL
set osd_rssi_alarm = 5
set osd_time_alarm = 0
set osd_alt_alarm = 1
set osd_dist_alarm = 0
set osd_neg_alt_alarm = 0
set osd_artificial_horizon_reverse_roll = FLOW_RAW
set osd_crosshairs_style = GAGAN
set osd_left_sidebar_scroll = VIBE
set osd_right_sidebar_scroll = NONE
set osd_sidebar_scroll_arrows = VIRTUAL
set osd_main_voltage_decimals = 0
set osd_attitude_angle_decimals = 75
set input_filtering_mode = FW_ALTITUDE

# profile
profile 1

set rc_expo = 60
set manual_rc_expo = 60

# battery_profile
battery_profile 1

set battery_capacity = 5000
set battery_capacity_warning = 750
set battery_capacity_critical = 350
```

shellixyz, 2018-06-20:
A number of your settings have wrong values. You should flash again with "full erase" turned on and modify again your settings. This is because you are using the development branch and is a side effect of the settings parameter groups versions not having being updated when the settings structures have been modified. We are aware that we need to fix that before the release of 2.0.

giacomo892, 2018-06-20:
Yeah.. 
``` inav_w_xy_res_v =  1000.000
 inav_w_acc_bias =  100.000```
Default values are 0.5 and 0.01

Also GPS altitude is not used if baro reading is sane as shown in your screenshot.

aubreyrjones, 2018-06-20:
I'm pretty reluctant to reset everything and manually set it up again. It took a lot of work in the first place, and you're telling me I can't even use the `diff` since those values are wrong.

I actually really hate this response. "Hi, I'm having trouble with my work."   "Well, first delete all your work."

If there is actually no way to reliably save settings, that should probably be documented somewhere. "Hey, write down everything on a napkin, because we have absolutely no meaningful way to save settings."

shellixyz, 2018-06-20:
@aubreyrjones You can use your diff. Just remove the values you didn't change from it.

shellixyz, 2018-06-20:
I also tested displaying the altitude in the configurator with a board on the bench without GPS. It is not consistent for me either. It doesn't display the same values as the baro using the default settings. When I move the board up/down it increases/decreases but whatever the height of the board when I stop moving it altitude goes back to 0.

aubreyrjones, 2018-06-20:
@shellixyz How would I know what I changed? That's the point of `diff`. The labels in the configurator have no bearing on the items in here. See my previous bug report about the "Mid Throttle" which is actually `rc_mid`.

Resetting the two `inav_w_*` values to defaults that @giacomo892 mentions reduces the magnitude of the error. It now oscillates with a value of only about 0.1m. I don't know if that's right. I can't find any documentation on how altitude is computed.

This is twice now that I've put in defect reports for things related to undocumented default settings. 

Where are any of these default values documented? Where is the feature `reset inav_w_*` that lets me reset only part of my configuration? Hell, where is the `default inav_w_*` command that even prints the default value? Where is *ANY* reliable way of saving settings between versions?

Because if one seriously has to brain wipe any craft they install a development version on, just so that they can be sure to meet the undocumented default values, why would anyone want to run development code? Now I know why so many top-level pilots recommend never updating a craft that works.

aubreyrjones, 2018-06-20:
I guess another question is, if changing these values can only do wrong, why do they even exist as configuration options? I understand it makes things convenient for developers during development, but each one of these configuration variables is a pinch point that will result in otherwise careful developers submitting doom-and-gloom defect reports.

giacomo892, 2018-06-20:
@aubreyrjones Copy the changes you did in a file. type defaults in the CLI e then restore and save only the good ones.

shellixyz, 2018-06-20:
The settings and defaults are documented there: https://github.com/iNavFlight/inav/blob/development/docs/Cli.md. I don't think there is a command to reset individual settings. 

You wouldn't have had this issue if you flashed development first with "full erase" then applied your diff from a previous version.

Also you are using the development branch you should be prepared for some breakage. There is a reason for it being called development. 

Personally I never used the `inav_w_*` settings but they are there to adapt to your hardware. You should not change them if you don't understand their meaning. What are you suggesting ?

shellixyz, 2018-06-20:
About my issue with altitude going back to 0 on the bench it is normal so if you see the same thing don't worry. The creator of the project (Digitalentity) told me the reason.

> The altitude is supposed to be pulled to zero on default settings when machine is disarmed so your observation is pretty normal.

closed 2018-09-02 by stale[bot]