iNavFlight/inav #3418: Soldering GPS and compass on Omnibus F4 V3

PackyFPV, 2018-06-22:
Hi there!!!
Today i received my new GPS that i would like to install on my flying Wing controlled by a Omnibus F4 V3 (https://www.banggood.com/Eachine-Wizard-X220S-FPV-Racer-Customized-Version-Omnibus-F4-V3-Betaflight-3_0_1-Flight-Controller-p-1177121.html?rmmds=myorder&cur_warehouse=CN) running Inav 1.9.1. It's about the entire day that i'm searching around for an answer to my problem, but even if it's a largely treated arguments, i wasn't able to find a solution of my problem; 
My GPS (https://www.banggood.com/UBLOX-NEO-M8N-BN-880-Flight-Control-GPS-Module-Dual-Module-Compass-p-971082.html?p=ZL241728738232015106&cur_warehouse=CN) has 6 wires (SCL,VCC,RX,TX,GND,SDA) and to connect them all to the board i wanted to make the following connections:

SCL GPS ------> SCL fc
VCC GPS ------> 5V fc
RX GPS ------> TX6 fc
TX GPS ------> RX6 fc
GND GPS ------> GND fc
SDA GPS ------> SDA fc

the problem is that on UART6 i have already enabled my sbus receiver and so, i cannot enable also the GPS on the UART6 (after rebooting the board the row in the port tab is reset). Some friends has suggested me to connect RX and TX of the GPS to TX3 and RX3 of the fc, but this means that i won't be able to connect SDA and SCL discarding the compass.
If there aren't any different options i'll choose the latter one... but i'm really ostinate and i hope you can help me to found a solution in order to keep everything of my gears!!!

Thank you very much for your help!!!
Best regards 
Packy

flounderscore, 2018-06-22:
On a fixed wing compass is optional (if not discouraged). I’d go with
option 2.

PackyFPV, 2018-06-22:
why it's discourage using a compass on fixed wing?

Artyum, 2018-06-23:
I have the same board. It's a limitation of it that I2C is shared with UART6, so you can't use it at the same time. If you want to use I2C you can try to use PPM for RX instead of SBUS or you should look for another board. But really compass for FW is not recommended. It's better to use GPS heading for nav calculations.

Tygrys-1, 2018-06-23:
AFAIK I2C2 is shared with UART3. If SBUS is on UART6, GPS can be on UART1. It was better arranged on V2 boards where SBUS was on UART1 and GPS could be used on UART6+SCL/SDA(shared with UART3) on one socket.

PackyFPV, 2018-06-23:
ok ok... if everyone said me that compass is not recommended on FW i trust it!!! i was searching for something like remapping or similar, but if it could give me trouble i prefer to give up!!! 
thank you all very much for your help!!! :)

Artyum, 2018-06-23:
@Tygrys-1 Of course I was wrong, sorry. I mean that I2C is shared with UART3. So UART3 can be used for GPS or Compass I2C.
For my mistake I can give easy hint for F4v3 on an airplane. The 4 PWMs connector can be removed to make space for servos output :)
[image]

closed 2018-07-03 by DzikuVx

Clawzy, 2019-01-02:
I’m having a related problem with this board. I moved my softserial from UART1 to UART3 to attach the GPS to the now free UART1. The GPS works fine but I can’t get softserial to work on UART3. I’d be interested in thoughts as to whether UART3 is not suitable for softserial.