iNavFlight/inav #3421: make toolchain installation like betaflight "make arm_sdk_install"

IllusionFpv, 2018-06-22:
It would be nice to have "make arm_sdk_install" like on betaflight, no need to add toolchain to PATH

stronnag, 2018-06-23:
not in my view, having an application trying to override the distro toolchain is highly undesirable.

stronnag, 2018-06-23:
OTOH, enabling an SDK install is a good idea, it's then defaulting to use an  obsolescent compiler that is where βF becomes annoying and counter-productive.
```
$ cd betaflight/
$ make
make/tools.mk:301: *** **ERROR** your arm-none-eabi-gcc is '8.1.0', but '7.2.1' is expected. Override with 'GCC_REQUIRED_VERSION' in make/local.mk or run 'make arm_sdk_install' to install the right version automatically in the tools folder of this repo.  Stop
```

So by all means, provide an easy means of installing an SDK, just don't enforce it people who know better.

IllusionFpv, 2018-06-23:
ok, I checked gcc version downloaded with install-toolchain.sh
```
Sviluppo/github/inav/gcc-arm-none-eabi-6-2017-q2-update/bin$ ./arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]
```
toolchain in betaflight is local and doesn't override global toolchain; a script can install appropriate version of toolchain(lastest for example), btw this is only a suggestion and not an real issue

closed 2018-09-05 by stale[bot]