iNavFlight/inav #3422: Disarm during flight (OMNIBUSF4V3 1.9.1)

Artyum, 2018-06-23:
Hi,
My FC disarmed when I was about to land. I lost control of aircraft. In OSD I saw the "flight summary" screen. After few seconds (~5) after resetting the radio and checking the sticks I regained controls but still in disarmed state, so without engine. I didn't have much time and I didn't noticed that I had autotrim mode enabled that protected FC to arm. So I landed safetly on OSD with that summary screen visible. I attached blackbox log and config dump. Unfortunately I can't read anything useful from it beside that ARM was OFF and log ends. That was kind of stressful but I 'm sure I didn't push the ARM stick by mistake.
[Logs.zip](https://github.com/iNavFlight/inav/files/2129815/Logs.zip)

DzikuVx, 2018-06-23:
Was it SBUS and on the radio failsafe configured as "no pulses"?

Artyum, 2018-06-23:
I try to figure out someting about the issue but In the meantime I would ask for explanation. At Fixed Wing Guide there is note about using an AUX switch for arming OR fixed_wing_auto_arm. In the CLI guide the note for fixed_wing_auto_arm says: "Requires enabled motorstop and NO arm switch configured." So switch arming should not be configured along with fixed_wing_auto_arm. In my case I have had both configured at the same time. Could it be possible cause of problem?
On the other hand, if we use switch arming with fixed_wing_auto_arm=OFF then is there another way to disable stick commands?
In my configuration I also had nav_disarm_on_landing = ON, could it be this function goes crazy?:)
As for my understanding the docs, it could be fine to use fixed_wing_auto_arm=ON with nav_disarm_on_landing=ON and stick arming not configured OR just a stick arming with the others OFF.

danarrib, 2018-06-26:
@Artyum L9R do have "NO PULSES" failsafe. To set it, you just power the L9R (but keep transmitter OFF). Then you shortly press F/S button on receiver.

Artyum, 2018-06-26:
I changed some serrings:
fixed_wing_auto_arm = OFF - I will use only aux switch arm but arming will be setup on a knob instead of switch. That way it will be impossible to disarm by accident. According to docs stick arming is disabled when using switch arming so it's like intended.
nav_disarm_on_landing = OFF - As I understand it applies only to automatic landing procedure and not for normal landing ex. in manual mode, but disabling it just in case.
nav_extra_arming_safety = OFF - I'm not sure if this makes sense, but I don't want the software to limit arming in event of GPS lost or not available.
And finally RX in "no-pulses" mode. Thanks @danarrib

closed 2018-09-08 by stale[bot]