iNavFlight/inav #3424: Quad in ALTHOLD mode jumps when hitting the ground

digitalentity, 2018-06-23:
This is a very old and very annoying issue. If a quad is in ALTHOLD mode and hits the ground hard enough it jumps for several meters before settling down. 

Cause of the issue is likely accelerometer saturating with rapid deceleration from impact.

No workaround exists at the moment.

giacomo892, 2018-06-23:
I've noticed that too. We haven't a limit in ACC weight during altitude estimation. 

This causes also altitude estimate going mad when a quad "soft" crash on ground. I've seen cases when estiamation drifted about +-100m until baro fusion brought it back to correct value.

wx4cb, 2018-07-03:
maybe if the last altitude was <= 1ft you could just ignore massive deceleration values?

teckel12, 2018-07-03:
@wx4cb That would only work if landing in the same location and the barometer is exactly the same as when launching (which is not typically the case).  I almost always show +/- 5 feet when landing in the exact same spot as launch.

fluffy, 2020-01-19:
When thinking about this, have a read of the "Barometer Ground Effect Compensation" section at https://docs.px4.io/v1.10/en/advanced_config/tuning_the_ecl_ekf.html 

That might be part of what is causing the issues here.

closed 2024-02-02 by DzikuVx