iNavFlight/inav #3425: LED indicator left/right don't work as intended (FW)

Pairan, 2018-06-24:
### Board and Version
* Matek F405-OSD
* iNav v1.9.1 (Problem exists for me since v1.6.1) / INAV/MATEKF405OSD 1.9.1 Apr 21 2018 / 13:26:36 (03a5c1922)
 
### Behavior / Problem
* LEDs are setup as 16 pieces on the back. 
* I've defined to let the outer 3 leds flash as indicator for LEFT and RIGHT turns
* When I turn left, only ONE led flashes! Turning right all 3 defined leds flash as intended
* I would have expected also to have 3 flashing leds on the left side while turning left
* A video showing the behaviour : https://youtu.be/sBecy6ZfKbg

# iNav Setup (LEDs)
led 0 15,15:E:GI:0
led 1 14,15:E:CTI:10
led 2 13,15:E:CTI:10
led 3 12,15:E:CT:10
led 4 11,15:E:CT:10
led 5 10,15:E:CT:10
led 6 9,15:E:CT:10
led 7 8,15:E:CT:10
led 8 7,15:W:CT:10
led 9 6,15:W:CT:10
led 10 5,15:W:CT:10
led 11 4,15:W:CT:10
led 12 3,15:W:CT:10
led 13 2,15:W:CTI:10
led 14 1,15:W:CTI:10
led 15 0,15:W:AIW:0

a complete DIFF is in the video description.

Pairan, 2018-08-23:
Still the problem is unsolved and not scheduled ... which actually is a pity

Pairan, 2018-08-23:
I tried to look into the LED.c and the header file but its far to abstract as I'm not aware of what is supposed to happen ...

I would like to help, but as it looks I'm the one that needs HELP here!

simtcrom, 2018-09-09:
Flight Controller - Omnibus F4 Pro v2 (clone)
LED Strip - CJMCU 8 Bit WS2812 8 Numbers.
Configuration - Led Number 1,2 and 6,8 are set color orange also selected indicators button on them. No other function defined on them.
Issue - When I move the roll stick to right or left, they Don't blink

Pairan, 2018-11-08:
Well ... doesnt seem to bother anybody somehow ...

How can I fix this myself?

simtcrom, 2018-11-08:
> Well ... doesnt seem to bother anybody somehow ...
> 
> How can I fix this myself?

Let me know if u find success.

teckel12, 2018-11-09:
I've experienced this for years with several different FC. I just don't use turn blinkers as they're not really useful anyway (I can't see them when 100 feet away or flying FPV).

closed 2019-01-22 by stale[bot]