iNavFlight/inav #3432: FF_PIKOF4OSD Servo3 not working

flounderscore, 2018-06-24:
### Board and Version
Furious FPV PikoF4 OSD running latest Development

### Behavior
Servo4 works, but Servo3 moved to "0" location

flounderscore, 2018-06-24:
I'm sorry. The posted the issue prematurely. Looks like Servo4 is stuck at 1ms output (looking on a scope). Servo3 is working correctly.

flounderscore, 2018-06-24:
Swapping the timer channel of output 2 and 4 solved my problem. That's the configuration that betaflight uses, too. I'm not sure what side effects that has on other things (esp. out 2).

```c
{ TIM2, IO_TAG(PA3),  TIM_Channel_4, TIM_EN, IOCFG_AF_PP, GPIO_AF_TIM2, TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR }, // S1_OUT // TIM2_CH4 | TIM5_CH4 | TIM9_CH2
        { TIM3, IO_TAG(PB1),  TIM_Channel_3, TIM_EN, IOCFG_AF_PP, GPIO_AF_TIM3, TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR }, // S2_OUT // TIM1_CH3N | TIM3_CH4 | TIM8_CH3N
        { TIM5, IO_TAG(PA2),  TIM_Channel_3, TIM_EN, IOCFG_AF_PP, GPIO_AF_TIM5, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO }, // S3_OUT // TIM2_CH3 | TIM5_CH3 | TIM9_CH1
        { TIM3, IO_TAG(PB0),  TIM_Channel_4, TIM_EN, IOCFG_AF_PP, GPIO_AF_TIM3, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO }, // S4_OUT // TIM1_CH2N | TIM3_CH3 | TIM8_CH2N
```

fiam, 2018-06-25:
@flounderscore That won't work unless you set the ESC to PWM 50hz because both S4 and S2 share timer 3. If no other option is available, you could set S2 and S4 as motors and S1 and S3 as servos.

flounderscore, 2018-06-25:
@fiam So if I understand you correctly, it should have worked before I swapped 2/4 and not afterwards? I'm using the default flying wing mixer, if that matters.

fiam, 2018-06-25:
@flounderscore Outputs declared as TIM_USE_FW_MOTOR can only share a timer with TIM_USE_FW_SERVO if the ESC protocol is PWM at 50Hz, otherwise it won't work properly.

I took a quick look at the target files and they seem incorrect to me, but I don't have this board to test.

flounderscore, 2018-06-26:
I think I understand what you're saying, but somehow I can only get servo4 to work if S4_out is on Channel_4 and the unused S2_out on Channel_3 (same as betaflight). Setting the ESC to 50Hz didn't help.

flounderscore, 2018-08-05:
By the way, this is still a problem for me unfortunately.

closed 2018-10-18 by stale[bot]

mirov, 2020-12-03:
A few years late, but I think your problem with S4 was that you were using Timer_3 and CH4, but that's not allowed for PB0 (it would have had to be T1_CH2 since T3_CH3 was already used).