iNavFlight/inav #3435: Add support for tbeacon gps output

JonnyFPV, 2018-06-24:
Hello, 

If it’s possible please add support for tbeacon 

Found here https://tbeacon.org/?page=20&language=en

It can read gps coordinates and speak it back over uhf :)

flounderscore, 2018-06-24:
Doesn't tbeacon just need to be connected to the GPS signal in parallel to the FC (GPS TX)?

wx4cb, 2018-06-27:
you need to be careful of parallelling serial ports between more than 2 devices because the open drain lines of one my drag down the other lines to the other device.

I would suggest just paralleling off the transmit line of the gps to the two devices and use the FC to configure the gps module itself.

shellixyz, 2018-07-03:
@JonnyFPV As @flounderscore and @wx4cb stated you only need to split the TX line from the GPS and connect it to the RX line of the tBeacon (don't forget the ground line too obviously). No support in iNav needed.

closed 2018-09-15 by stale[bot]