iNavFlight/inav #3438: F3 EVO default compass direction of MPU9250 is not correct

zipray, 2018-06-25:
F3 EVO FC default compass direction of MPU9250 is not correct in Inav 1.9, and it must be set to flip90CW.And the accelerometer, the gyro is not needed

fiam, 2018-06-25:
@zipray Give this hex a try, please. Let me know if it works properly with the default mag alignment. I did fix a big in the driver and it seems to me the default orientation should now be CW_270 (without flip), but I don't have this board to test it.

[inav_2.0.0_SPRACINGF3EVO_TEST_3434.hex.zip](https://github.com/iNavFlight/inav/files/2134256/inav_2.0.0_SPRACINGF3EVO_TEST_3434.hex.zip)

fiam, 2018-06-25:
Probably related to #3434

zipray, 2018-07-07:
Thank you very much for your reply.I updated the firmware.Although it can show the right direction in the GUI configuration, it cannot do a complete accelerometer calibration.Step 3 of the accelerometer calibration will show that step 4 has been completed, and the calibration of step 3 cannot be completed regardless of positioning.In addition, I've updated firmware 2.0 RC2, which can be calibrated, but like version 1.9.1, you must configure the compass as CW90flip.The compass deviation in flight becomes unstable.

fiam, 2018-07-16:
Fixed in #3612, we’ll fix the Z axis in 2.1

commit 421d27e8 (2018-06-25): Fix orientation of the drivers for AK8963, AK8975 and MPU9250 MAGs

Due to the unsual axis definition of the AKxxxx MAGs (the +Y axis
is 90ºCW from the +X axis when +Z points up), the driver needs to
invert e.g. X and Z, then rotate rather than inverting all of
X, Y and Z like it was doing.

It seems this bug was introduced in 3d024c6f3 back in November,
between 1.8 and 1.9 and some targets were written or updated in the
meantime assuming the driver was correct and thus have incorrect mag
alignments. I've tried to fix them all, but I don't have all these
boards and while ones with the MP9250 a [... cut]
PR #3612 (2018-07-16): Fix default compass direction for SPRACINGF3EVO
Fixes #3438
commit 11cf23f2 (2018-07-16): Fix default compass direction for SPRACINGF3EVO

Fixes #3438
PR #3447 (2018-08-10): Fix orientation of the drivers for AK8963, AK8975 and MPU9250 MAGs
Due to the unsual axis definition of the AKxxxx MAGs (the +Y axis
is 90ºCW from the +X axis when +Z points up), the driver needs to
invert e.g. X and Z, then rotate rather than inverting all of
X, Y and Z like it was doing.

It seems this bug was introduced in 3d024c6f3 back in November,
between 1.8 and 1.9 and some targets were written or updated in the
meantime assuming the driver was correct and thus have incorrect mag
alignments. I've tried to fix them all, but I don't have all these
boards and while ones with the MP9250 ar [... cut]
closed 2018-07-16 by fiam