iNavFlight/inav #3441: Firmware Request for DALRC F405 AIO

sammichael25, 2018-06-25:
[https://www.banggood.com/DALRC-F405-AIO-Betaflight-F4-Flight-Controller-BEC-OSD-50A-Blheli-32-3-8S-Brushless-ESC-for-RC-Drone-p-1300573.html?rmmds=search&cur_warehouse=CN](url)

eroc1969, 2018-09-13:
Is this available or coming soon or did I miss it? Thanks for all your hard work.

sammichael25, 2018-09-13:
It’s released, check the releases page for the hex

stellarhopper, 2018-09-19:
It looks like this was added, but then after 2.0.0-rc4, it stopped getting included in the releases.
I see the source files for the target are still there - is it just a build/automation issue?

stellarhopper, 2018-09-19:
@eroc1969  Here is a 2.0.0 local build I spun up for this board, until the automation gets fixed.

https://stellarhopper.com/owncloud/index.php/s/nDSBAoRH9sba2Sh

eroc1969, 2018-09-21:
Aio and regular f405. I was needing the regular  Dalrc f405 NOT the aio. So I can now download the hex and it's good to go. Thank you everyone for your help on this. This newbie is still trying to figure out my matek f722 baro BMP 280 i2c conflict. Is pull up resistors the final decision. New to all this.