iNavFlight/inav #3442: Beitian BN-880 extremely slow

golffer, 2018-06-25:
iNav 1.9.3
Omnibus F4 V3
Frsky XSR, SBus
Beitian BN-880, UART 1, stock without any setup or config
Compass, I2C
Mode setup:NAV ALTHOLD, NAV POSHOLD, NAV RTH

Issue1:  
Beitian 880 is just linked by cable, keeping away from XSR or Battery cable.
Sometime, iNav just show Waiting for GPS 3D Fix, can not find satellite (near window).
Sometimes, it takes more than 15 minutes to find satellite.

Issue2:
Even it finds 5~7 satellite (satellite q'ty jump between 5, 6, 7), still can not arm (FC blue LED is flashing rapidly with beep sound).

May someone teach me how to solve?
Thanks

teckel12, 2018-06-25:
I have two Beitian BN-880 and neither will ever get a satellite lock inside my home, even near a window.  It must be outside.  Even outside if it's too close to my house it won't get a lock (or will take FOREVER).

Also, even if it has 6 satellite locks, INAV may still not allow you to fly.  It also needs a good 3D fix with a minimum HDOP accuracy.

Finally, once it gets a lock it stores the positions of the satellites.  If you fly a second time, it will be MUCH quicker to get a GPS lock.  Once you get a good lock, you could then take it into your house and by a window it will keep a GPS lock.  That's because it has already downloaded the satellite positions.

In any case, try it in the field and you should get different results.

golffer, 2018-06-26:
Hi, Tecken12

Thanks for comment. 
Should we setup BN-880 in U-Center?
Some people mention they can get quick satellite locked.
I envy them.

teckel12, 2018-06-26:
The way GPS works is that the receiver can "hear" the GPS satellite signals right away. However, it has no idea where they are nor what direction they're heading. This information is downloaded to the GPS receiver, and this download isn't very fast. Once it knows the position and the direction of the satellites, it saves that information to non-volatile memory. When you fly a second time, GPS lock takes just a couple seconds, as it stored the satellite positions.

You're just not giving your GPS receiver time to do what it needs to. So it appears it's slow when it's not. In the case of your phone, for example, it downloads via the internet connection the locations and trajectories of all GPS satellites. So, when it first "hears" a GPS satellite it doesn't need to wait. Your quad doesn't have an Internet connection to pre-download the satellite positions, so it can't be as fast.

Your GPS isn't any slower than anyone else's, they're just giving it open sky to download location information.

roopej, 2018-06-26:
While it does not help with your problem, just for the record - my BN-880 has never taken more than 60 seconds to get lock no matter if started outside or inside the house (by the window). Maybe I’ve just been lucky or maybe some units are more sensitive than others. Mostly I have been using Galileo.

Lately I have been experimenting with Radiolink TS100 unit. So far I would say it is locking somewhat faster than BN-880. Maybe you could compare if you would get better results with different model?

shellixyz, 2018-06-26:
Maybe the battery on your BN-880 is dead ?

davidngrc, 2018-06-28:
is there an API to check the battery on BN-880 or other gps device? so that we can see it on the iNav configuration?

teckel12, 2018-06-28:
@davidngrc I don't think the BN-880 reports the battery voltage to be able to display that information.  The battery isn't critical, the GPS will work without it, it will just take longer to get a lock between flights.

golffer, 2018-07-03:
Hey, Guys
I use FTDI to connect BN-880 to U-Center.
It works well and fast. So BN-880 is no problem.
But, when I install 880 to FC, it takes a long time waiting.

UART3 and I2C share the same pin.
Should I disable UART3 to let I2C works fast?
How to do it in CLI?
Thanks

r3df1v3, 2018-07-07:
Hi. I have a similar issue with a Beitian BN-220 and Omnibus F4 Pro (ledstrip M5), iNav V. 2.0 RC2,
Sometimes, the GPS is not detected (not the satellites signal, but the GPS Module itself). Blue led blinks and buzzer beeps. In Configurator, GPS icon appears red.
Sometimes it  works fine for some minutes, and then it fails, and then after a while, it works again. This  happens while fliying or on the bench.

teckel12, 2018-07-08:
@golffer Yes, you must disable UART 3 when using i2c. Just don't select anything for UART 3 in the configurator.

teckel12, 2018-07-08:
@r3df1v3 I've had this happen many times due to the JST connector being slightly out on the GPS or FC side. When GPS is reporting an error, push in both connectors and restart. This has ALWAYS restored GPS for me. This is also why people hate JST connectors.

r3df1v3, 2018-07-11:
Hi. I've tested the conectivity from the pads on the board, to the pads on the GPS with a mutimeter, and everything is OK (all connectors, and cables). I've tried two different M8N GPS modules, and the random disconnection still happened.
However, If I connect the GPS in UART3 instead of UART1, it seems to work fine. Also, I get 8-12Hz using UBLOX7 on this port. I' haven't tested it yet on the field.

r3df1v3, 2018-07-11:
I'm using a Thomas Scherrer Long Range UHF RX, using PPM, on the 1st pin, just above the PWM outputs.

fiam, 2018-07-11:
@r3df1v3 That should not cause any issues with UART1 nor the GPS, as long as you're not putting the UHF RX next to the GPS :-).

r3df1v3, 2018-07-11:
The aircraft is a Multiplex Easy Glider Pro. RX antennas are on the vertical stab, and gps module is embedded in the foam near the tail. The RX should no affect the GPS, as it does not transmit any signal.
The video TX, on the other hand, is a Fatshark 250mW @1280MHz. It Uses a SPW antenna (Sircana's), on the canopy, so it's almost 1 meter from the VTX to the GPS.
Also, i've tried the radio control TX (Scherrer 2 Watts) near the GPS module, and it still picks up 7 satellites inside home, by the window...

closed 2018-09-23 by stale[bot]