iNavFlight/inav #3445: Quad flying into random directions while activating Nav pos hold or RTH mode

ninja-zx11, 2018-06-25:
Hi,
       Just installed INAV 1.9.1 on one of my quad which was running betaflight.FC board is Flip32 F4 running airbotf4 target.M8n GPS puck on a mast away from the electronics and Aluminum foiul under the GPS.I calibrated properly as per inav documents and on my first test,everything worked good.Nav pos hold was solid,no toilet bowl effect and RTH worked perfect.

But on my second test(second battery),after flying FPV for few minutes,i decided to test 3D holdbut it didn't work well this time.I was getting around 23 sats.What was happening is that quad was trying to flying into random directions after turning on 3D hold mode.I remember,once it tried to fly towards east and second time ,it tried to fly towards west.Altitude hold worked perfect all the times.I have an external MS5611 barometer covered with open cell foam.I have selected 5" GPS quad preset from the inav configurator.

Weird is that it didn't show that problem after.I flew 3-4 batteries after and it was working fine.

My Diff:
`# diff

# version
# INAV/AIRBOTF4 1.9.1 Apr 21 2018 / 13:06:13 (03a5c1922)

# resources

# mixer

# servo mix

# servo

# feature
feature SOFTSERIAL
feature GPS
feature TELEMETRY
feature CURRENT_METER
feature PWM_OUTPUT_ENABLE

# beeper

# map
map TAER

# name

# serial
serial 0 64 115200 38400 0 115200
serial 5 2 115200 57600 0 115200
serial 30 32 115200 38400 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1900 2100
aux 1 1 1 1300 1700
aux 2 2 1 1800 2100
aux 3 3 4 1300 2100
aux 4 9 4 1850 2100
aux 5 8 3 1300 2100
aux 6 17 2 1300 2100

# adjrange

# rxrange

# master
set looptime = 2000
set gyro_sync = ON
set align_gyro = CW0FLIP
set gyro_hardware_lpf = 188HZ
set gyro_lpf_hz = 90
set gyro_notch1_hz = 200
set gyro_notch1_cutoff = 100
set align_acc = CW0FLIP
set acc_hardware = MPU6000
set acczero_x = 160
set acczero_y = -48
set acczero_z = -983
set accgain_x = 4099
set accgain_y = 4076
set accgain_z = 4028
set align_mag = CW270FLIP
set mag_hardware = HMC5883
set mag_declination = 1610
set magzero_x = 83
set magzero_y = -20
set magzero_z = -13
set baro_hardware = MS5611
set pitot_hardware = NONE
set receiver_type = SERIAL
set serialrx_provider = SBUS
set min_throttle = 1040
set max_throttle = 1900
set motor_pwm_rate = 1000
set motor_pwm_protocol = ONESHOT125
set failsafe_procedure = RTH
set vbat_scale = 1090
set vbat_max_cell_voltage = 420
set battery_capacity = 1500
set battery_capacity_warning = 450
set battery_capacity_critical = 300
set current_meter_scale = 320
set current_meter_offset = 6
set gps_sbas_mode = WAAS
set gps_ublox_use_galileo = ON
set smartport_fuel_unit = PERCENT

# profile
profile 1

set mc_p_pitch = 58
set mc_i_pitch = 50
set mc_d_pitch = 22
set mc_p_roll = 43
set mc_i_roll = 40
set mc_d_roll = 20
set mc_p_yaw = 70
set dterm_lpf_hz = 80
set dterm_notch_hz = 200
set dterm_notch_cutoff = 100
set roll_rate = 50
set pitch_rate = 50
set yaw_rate = 45

#
[`]
[blackbox_log_2018-06-23_153625.zip](https://github.com/iNavFlight/inav/files/2134120/blackbox_log_2018-06-23_153625.zip)

tgreer, 2018-06-25:
Is your compass orientated correctly? Sounds like it's not. Take it outside and check with a compass, including whilst holding the craft tilted forward/backwards/left/right.

ninja-zx11, 2018-06-25:
Yes its oriented correctly and it shows east as east.I used CW270 flip because its mounted upside down inside a bangood M8n GPS puck.Also compass reading stays same even if i lift any side.(reading stays within 1~2 degree).

ninja-zx11, 2018-06-25:
Sure i will check and report back.Thanks.

ninja-zx11, 2018-06-26:
I checked my compass and it seems to give proper readings.Looking at the quad from top and rotating it clockwise,reading increases like 0-90-180-270-360.
Reading also stays within 1~2 degrees whilst tilting the craft  forward/backwards/left/right.

DzikuVx, 2018-06-26:
What you describe is in 95% of the cases the result of magnetometer problems:
https://www.youtube.com/watch?v=FlEm0-pXNf0

ninja-zx11, 2018-06-26:
Thanks DzikuVx for the reply.But my quad holds 3D-position beautifully even in a windy situation.No toilet bowling at all.I have calibrated it as per the Inav documents and its monted high on a mast  away from all the electronics.I have even installed aluminium foil under-neath the compass board.

wilco1967, 2018-06-26:
Could it be related to how you activate poshold ?
If your RTH works fine, I think you can conclude your compass is fine.

Try doing a RTH, and then while it is coming back, switch to poshold without going to any other mode in between.... 

Not sure, but Perhaps this is similar to what I noticed in issue 3448

ninja-zx11, 2018-06-28:
Sure i will try that.Thanks

ninja-zx11, 2018-06-28:
Is there any need to enter the magnetic declination value manually on F4 board?I have entered it manually.Does it make any difference?

stronnag, 2018-06-28:
There is no need to enter it, it is automatically provided. Unless you live in some insane magnetic anomaly, the difference with manually provided value is too small to be significant.

ninja-zx11, 2018-06-28:
Okay Thanks.
One more thing i found yesterday.
My GPS mast can be tilted to any position from 0 to 90 degrees but also can be locked either at 90 degrees or 0 deg.Yesterday i tilted it by approx 20 deg thinking that GPD would point up at 90 deg while the quad is in forward motion due to its tilt.My quad was behaving exactly like when i noticed on my second problematic flight.I think tilting compass chip back disturbed some readings and quad was flying into random directions.
But i am 100% sure on my first problematic flight,GPS was locked at in upright position at 90 degrees.

Any chance that problem i faced before was due to activating NAV Pos hold while the quad was in forward motion and was tilted towards front quite a bit ?And at that tilted position my compass's heading readings were messed up?

tgreer, 2018-06-28:
I wouldn't think so. I hit pos hold whilst full pelting forwards and it's
completely fine.

ninja-zx11, 2018-06-30:
Just traced my I2C bus containing HMC5883 and MS5611.Trying to rule out any I2C bus problem.I get 0 I2C errors if looking at Inav configurator.Could anybody please have a look at traces if i need to tweak the pull up resistors.I didn't install any pull ups.All default.
2 traces are from the clock line and rest of them are from Data line at different zoom levels.Thanks

[image]
[image]
[image]
[image]
[image]
[image]
[image]

closed 2018-09-12 by stale[bot]