iNavFlight/inav #3448: Poshold not always engaging in iNav 2.0.0 Dev

wilco1967, 2018-06-25:
### Board and Version
INAV/MATEKF722 2.0.0 Jun 24 2018 / 19:29:34 (90e0c4be) 
Custom compiled Development version from yesterday

### Behavior
Flying iNav 2.0.0 development on a mini Talon airplane.

Problem is, Poshold does not consistently engage when commanded.

When coming out of 3CRS (3d stableisation mode, or 'cruise mode'), going into poshold, it keeps flying in a straight line, as if still in 3CRS, but I cannot control it using the stick. 
The OSD tells me it is in fact in HOLD mode, so the command is received, and 'understood'..... but it keeps flying in a straight line

However, if I come from RTH mode, and then switch to poshold, it works just fine, and it start circling....
Also, when coming from ANGL mode, to poshold, it also works just fine....

The OSD shows the requested mode alright, but it doesn't respond as HOLD....

I do ALL my mode switching using a single RC channel (CH5), using a mixer in Taranis. In the mode's tab, I have the necessary combinations set so it performs the required action.
As a result, the same combination of modes will always be set, regardless of how I select it , and from where it comes (as the Taranis can only output a single value at any moment ).

For example, i have RTH set up as 2000 uS (on CH5).
CH5 at 2000 uS activates:  nav-RTH and turn assist (I know, not required, but nevertheless)... works just fine

Poshold is set up as 1200 uS, which activates NAV poshold, NAV Althold, Angle, turn assist, Alt hold.... sometimes this works, and sometimes it doesn't (but it's always the same 1200 uS)

auto-level + althold + cruise  I have at 1400 uS, which activates Angle, turn assist, alt-hold, heading hold, cruise... always works....

So when going from 1400 (auto-level+althold + cruise), going to 1200 (poshold),, it keeps flying straight, even though the OSD tells me it IS in HOLD

So when going from 2000 (RTH) to 1200 (POSHOLD), or from 1100 to 1200 (simple angle mode) it works just fine....

GPS works just fine all the time..... 

here's a DIFF
[https://pastebin.com/H2xgPW93](url)

Starting from about 6:40 minutes into the video (19:46 GPS time), I do a couple of switching actions, where you see the behaviour.... I (hardly) touch the sticks during these trials.... just changing modes.
just watch the mode indicator on OSD, and see how it behaves....

link to video
[https://drive.google.com/open?id=17IfjCQ3zA2uC1YPtgF-4bYmkzkaFP02F](url)

here's the blackbox log
action happens starting around 6:32 into the log
[https://drive.google.com/open?id=1cWOvn6czlsjW13h_01rjBRaeAm4kTYEP](url)

Poshold is not something I often use, but I do fly a lot of WP missions, and the new cruise mode, which is great..... except that funny poshold, the rest works perfect !
Thanks a lot !

stronnag, 2018-06-25:
FYI, mode changes. Note interaction with CRUISE mode.

```
$ inav_modes.rb ~/dl/LOG00012.TXT 
iNav version = 2.0.0 (states eq 2.0.0)
Time(s)	Elapsed(s)	State
 127.0	(   0.0)	nav_state_launch_wait (26)
 131.0	(   4.0)	nav_state_launch_in_progress (28)
 135.7	(   8.7)	nav_state_idle (1)
 154.0	(  27.0)	nav_state_cruise_3d_in_progress (33)
 202.8	(  75.8)	nav_state_poshold_3d_in_progress (7)
 202.8	(  75.8)	nav_state_rth_head_home (10)
 209.6	(  82.6)	nav_state_poshold_3d_in_progress (7)
 211.2	(  84.2)	nav_state_cruise_3d_in_progress (33)
 233.1	( 106.0)	nav_state_cruise_3d_adjusting (34)
 238.8	( 111.8)	nav_state_cruise_3d_in_progress (33)
 242.9	( 115.9)	nav_state_cruise_3d_adjusting (34)
 244.6	( 117.6)	nav_state_cruise_3d_in_progress (33)
 289.1	( 162.1)	nav_state_cruise_3d_adjusting (34)
 297.0	( 169.9)	nav_state_cruise_3d_in_progress (33)
 299.3	( 172.3)	nav_state_cruise_3d_adjusting (34)
 301.3	( 174.3)	nav_state_cruise_3d_in_progress (33)
 302.3	( 175.3)	nav_state_cruise_3d_adjusting (34)
 309.8	( 182.8)	nav_state_cruise_3d_in_progress (33)
 320.6	( 193.6)	nav_state_cruise_3d_adjusting (34)
 323.1	( 196.1)	nav_state_cruise_3d_in_progress (33)
 326.8	( 199.8)	nav_state_cruise_3d_adjusting (34)
 329.1	( 202.0)	nav_state_cruise_3d_in_progress (33)
 334.7	( 207.7)	nav_state_cruise_3d_adjusting (34)
 336.8	( 209.8)	nav_state_cruise_3d_in_progress (33)
 377.7	( 250.7)	nav_state_cruise_3d_adjusting (34)
 380.1	( 253.1)	nav_state_cruise_3d_in_progress (33)
 410.2	( 283.1)	nav_state_poshold_3d_in_progress (7)
 410.2	( 283.2)	nav_state_rth_head_home (10)
 519.7	( 392.7)	nav_state_poshold_3d_in_progress (7)
 519.7	( 392.7)	nav_state_cruise_3d_in_progress (33)
 536.8	( 409.8)	nav_state_poshold_3d_in_progress (7)
 536.8	( 409.8)	nav_state_rth_head_home (10)
 540.3	( 413.3)	nav_state_poshold_3d_in_progress (7)
 540.3	( 413.3)	nav_state_cruise_3d_in_progress (33)
 543.4	( 416.4)	nav_state_poshold_3d_in_progress (7)
 549.3	( 422.3)	nav_state_cruise_3d_in_progress (33)
 550.0	( 422.9)	nav_state_poshold_3d_in_progress (7)
 551.9	( 424.9)	nav_state_rth_head_home (10)
 553.6	( 426.6)	nav_state_poshold_3d_in_progress (7)
 557.2	( 430.2)	nav_state_cruise_3d_in_progress (33)
 558.5	( 431.5)	nav_state_poshold_3d_in_progress (7)
 561.3	( 434.3)	nav_state_cruise_3d_in_progress (33)
 564.5	( 437.5)	nav_state_poshold_3d_in_progress (7)
 584.3	( 457.3)	nav_state_cruise_3d_in_progress (33)
 585.9	( 458.9)	nav_state_cruise_3d_adjusting (34)
 605.3	( 478.3)	nav_state_cruise_3d_in_progress (33)
 629.8	( 502.8)	nav_state_poshold_3d_in_progress (7)
 629.8	( 502.8)	nav_state_rth_head_home (10)
 635.4	( 508.4)	nav_state_poshold_3d_in_progress (7)
 640.2	( 513.2)	nav_state_cruise_3d_in_progress (33)
 643.4	( 516.4)	nav_state_poshold_3d_in_progress (7)
 656.5	( 529.4)	nav_state_rth_head_home (10)
 657.9	( 530.9)	nav_state_poshold_3d_in_progress (7)
 662.2	( 535.2)	nav_state_cruise_3d_in_progress (33)
 663.1	( 536.1)	nav_state_poshold_3d_in_progress (7)
 664.2	( 537.2)	nav_state_rth_head_home (10)
 665.4	( 538.4)	nav_state_poshold_3d_in_progress (7)
 667.7	( 540.7)	nav_state_rth_head_home (10)
 677.4	( 550.4)	nav_state_poshold_3d_in_progress (7)
 679.1	( 552.1)	nav_state_cruise_3d_in_progress (33)
 680.7	( 553.7)	nav_state_poshold_3d_in_progress (7)
 682.1	( 555.1)	nav_state_cruise_3d_in_progress (33)
 683.6	( 556.6)	nav_state_poshold_3d_in_progress (7)
 685.8	( 558.8)	nav_state_rth_head_home (10)
 687.1	( 560.0)	nav_state_poshold_3d_in_progress (7)
 688.7	( 561.7)	nav_state_cruise_3d_in_progress (33)
 690.0	( 563.0)	nav_state_poshold_3d_in_progress (7)
 691.2	( 564.2)	nav_state_cruise_3d_in_progress (33)
 692.7	( 565.7)	nav_state_poshold_3d_in_progress (7)
 694.3	( 567.3)	nav_state_rth_head_home (10)
 786.2	( 659.2)	nav_state_poshold_3d_in_progress (7)
 786.3	( 659.3)	nav_state_cruise_3d_in_progress (33)
 787.3	( 660.3)	nav_state_poshold_3d_in_progress (7)
 788.7	( 661.7)	nav_state_cruise_3d_in_progress (33)
 789.3	( 662.3)	nav_state_poshold_3d_in_progress (7)
 789.4	( 662.4)	nav_state_rth_head_home (10)
 790.4	( 663.4)	nav_state_poshold_3d_in_progress (7)
 792.0	( 664.9)	nav_state_cruise_3d_in_progress (33)
 793.7	( 666.7)	nav_state_poshold_3d_in_progress (7)
 795.7	( 668.7)	nav_state_cruise_3d_in_progress (33)
 796.9	( 669.9)	nav_state_poshold_3d_in_progress (7)
 798.3	( 671.3)	nav_state_rth_head_home (10)
 799.8	( 672.8)	nav_state_poshold_3d_in_progress (7)
 800.8	( 673.8)	nav_state_cruise_3d_in_progress (33)
 802.6	( 675.6)	nav_state_poshold_3d_in_progress (7)
 812.0	( 685.0)	nav_state_cruise_3d_in_progress (33)
 813.2	( 686.1)	nav_state_poshold_3d_in_progress (7)
 813.2	( 686.2)	nav_state_rth_head_home (10)
 814.6	( 687.6)	nav_state_poshold_3d_in_progress (7)
 817.6	( 690.6)	nav_state_cruise_3d_in_progress (33)
 818.7	( 691.7)	nav_state_idle (1)
 823.2	( 696.2)	nav_state_poshold_3d_in_progress (7)
 824.7	( 697.7)	nav_state_idle (1)
 828.2	( 701.2)	nav_state_poshold_3d_in_progress (7)
 829.9	( 702.9)	nav_state_idle (1)
 831.4	( 704.4)	nav_state_poshold_3d_in_progress (7)
 833.9	( 706.9)	nav_state_cruise_3d_in_progress (33)
 835.5	( 708.5)	nav_state_poshold_3d_in_progress (7)
 836.8	( 709.8)	nav_state_rth_head_home (10)
 847.7	( 720.7)	nav_state_poshold_3d_in_progress (7)
 847.8	( 720.8)	nav_state_cruise_3d_in_progress (33)
 851.1	( 724.1)	nav_state_cruise_3d_adjusting (34)
 851.3	( 724.3)	nav_state_cruise_3d_in_progress (33)
 851.4	( 724.4)	nav_state_cruise_3d_adjusting (34)
 852.5	( 725.5)	nav_state_cruise_3d_in_progress (33)
 856.5	( 729.4)	nav_state_idle (1)
 901.6	( 774.5)	nav_state_rth_head_home (10)
 920.4	( 793.4)	nav_state_idle (1)
 949.0	( 822.0)	nav_state_cruise_3d_in_progress (33)
 949.5	( 822.4)	nav_state_idle (1)
```

shellixyz, 2018-06-26:
@wilco1967 Thank you for the detailed report. Will be fixed in the final 2.0 version.

PR #3458 (2018-06-27): Fix POSHOLD mode
Tested. Fixes #3448
closed 2018-06-27 by DzikuVx