iNavFlight/inav #3449: osd_crosshairs_style and sidebar_scroll feedback

Artyum, 2018-06-25:
As for the experimental settings:
set osd_crosshairs_style = AIRCRAFT
set osd_left_sidebar_scroll = GROUND_SPEED
set osd_right_sidebar_scroll = ALTITUDE
For me an aircraft pilot these are nice features. Few remarks: The aircraft crosshair style covers the horizon line that there is two dots on left side and one on right side (screen attached). Would be nice to make it symmetrical :) 
[image]
The sidebar scrool seems to work but is not very immersive. I think that at some speed ratio due to low vertical res it's hard to distinguish if it's going up or down. Despite that I will leave it enabled.

fiam, 2018-07-10:
@Artyum Note that you can use `osd_sidebar_scroll_arrows` to show the scrolling direction. I'm adding these to the configurator for 2.0, they should be pretty easy with the new settings  MSP API.

fiam, 2018-07-10:
Fixed by https://github.com/iNavFlight/inav-configurator/pull/533

closed 2018-07-10 by fiam