iNavFlight/inav #3453: Issues to be sorted out before 2.0 final release

giacomo892, 2018-06-26:
1.  Name appears in diff even when it has its default value. 

[image]

Artyum, 2018-06-27:
Configurator 2.0.0.RC1 Some minor things:
1) In GUI OSD the "Sag Compensated Battery Cell Voltage" and "Fw Min Throttle Down Pitch Angle" control the same icon in GUI.
2) In Mixers panel for Airplane, the servo mixers for flaps are added despite "Has flaps" is disabled.

DzikuVx, 2018-06-27:
@Artyum please create new tickets in Configurator project for your findings. This way we can keep up what goes where. And one ticket for one problem please. Thanks