iNavFlight/inav #3456: INAV 2.0 RC1 - FrSky Telemetry

ThomasH3, 2018-06-26:
hello ... for frsky telemetry (Fport)
on V1.9 I selected .... FPort, Serial_inverted on, Serial_halfduplex on, telemetry_inverted on, telemetry halfduplex on

on V2.0 I can't receive telemetry data .... could it be reason that telemetry_halfduplex can't be changed ?

DzikuVx, 2018-06-26:
in 2.0 meaning of `inverted` changed. Now it means "inverted comparing to protocol default"

fiam, 2018-06-26:
Minor nitpick: It was changed in 1.9.1. See the release notes for the documentation https://github.com/iNavFlight/inav/wiki/1.9.1-Release-notes

ThomasH3, 2018-06-26:
How can i get telemetry via fport on inav ? I' m using an spf3neo with xsr receiver.

fiam, 2018-06-26:
@ThomasH3 Select FPort as the RX protocol in the configuration tab and enter the following CLI commands:

```
set serialrx_halfduplex = ON
set serialrx_inverted = ON
```

Unfortunately, FrSky has specified the protocol in one way (non inverted, full duplex) but the current generation of receivers implements it in other way (inverted, half duplex). That's why it doesn't work without those CLI commands. Once the dust settles, we'll make it easier.

ThomasH3, 2018-06-27:
I've tried that ...... I can't receive telemetry data.

ThomasH3, 2018-06-27:
CLI setting is
serialrx_provider = FPORT
serialrx_inverted = on
serialrx_halfduplex = on
telemetry_inverted = off
**receiver_type = SERIAL**

thanks !!

closed 2018-06-27 by ThomasH3

fiam, 2018-06-27:
@ThomasH3 That's why I told you to select FPort in the configuration tab, because that implies `receiver_type = SERIAL` and `serialrx_provider = FPORT`.