iNavFlight/inav #3462: Stuck with "Waiting for data" in Servos and Mixer, CLI is ok

Top97, 2018-06-27:
Hello,

Strange problem with iNav config 2.0.0 RC1
I have Omnibus F4 Pro V2 flashed with 1.9.1

When click on Mixer or Servos - the iNav stuck with the animation of "Waiting for data" endless and can't change to other section anymore. I need to disconnect and connect again. 

Using CLI to change Servos or adjust Mixing - working good, but still can't use the GUI

Thank you !

Roofneck, 2018-06-27:
Same here stuck on "waiting for data" mixer and servos. Omnibus F3 Pro 1.9.1
Windows 10.

DzikuVx, 2018-06-27:
Please do not use Configurator 2.0 with older firmwares

giacomo892, 2018-06-27:
@Top97 Just use https://github.com/iNavFlight/inav-configurator/releases/tag/1.9.3 for INAV 1.9.1 but I suggest you to upgrade to 2.0RC1 :)

fiam, 2018-06-27:
We used to have a "pretty" error message in those tabs telling users to upgrade their firmware, I guess we should fix that.

closed 2018-06-27 by DzikuVx