iNavFlight/inav #3463: Request support for Betaflight F7

TheRealOrange, 2018-06-27:
This is a feature request. I would like to use the iNav firmware on the relatively new Betaflight F7 board. I know some other boards also have dual gyroscope, but I purchased this before finding out about iNav. If anyone else has custom compiled some firmware, please tell me how. I realise Betaflight is different from iNav and I already purchased the Betaflight board but I am interested in GPS features such as return to home. The supported boards page stated that most F7 boards were supported. I am sorry for the trouble and thank you.

TheRealOrange, 2018-07-21:
I believe it may be possible to look at the timer channels to motor assigned in the betaflight target file and then look at iNav target file syntax and create a custom target file that dictates the timers specific to the motor outputs.

closed 2018-10-03 by stale[bot]

mchaos, 2018-10-17:
I too would like to see betaflightF7 supported...