iNavFlight/inav #3470: Stuff to note when upgrading to INAV 2.0.0

teckel12, 2018-06-28:
Converting from 1.9.1 to 2.0.0 the following changes should be noted:

* Full chip erase on flash?
* Mixer must be configured
* Must re-calibrate
* All OSD settings redone
* CLI settings that have been deprecated:
    * feature -RX_PPM
    * feature -SERVO_TILT
    * feature -CHANNEL_FORWARDING
    * set mid_rc
    * set gimbal_mode
    * set iterm_ignore_threshold
    * set yaw_iterm_ignore_threshold
* Invalid value ranges:
    * set 3d_deadband_throttle = 50 (Allowed range: 1000 - 2000)
    * set fw_autotune_ff_to_i_tc = 600 (Allowed range: 1000 - 5000)

To get INAV 2.0.0 to fly at all, I needed to reset the mixer and re-calibrate.  Until I did those two things, I could not arm.  The rest are notes (maybe for some no mention needs to be made) but those were all the errors when trying to load the 1.9.1 config settings in 2.0.0.

leeph, 2018-07-23:
Why the hell is RX_PPM deprecated? It's still available as a selectable option. Lots of people still use PPM RXs - especially long-range UHF stuff.

teckel12, 2018-07-24:
It could just be renamed or done a different way. Only the setting was removed.

shellixyz, 2018-07-25:
@leeph PPM is still supported: `set receiver_type = PPM`

c4xp, 2018-07-27:
I was going to use SERVO_TILT and CHANNEL_FORWARDING for my long-range UAV. Is this going away for good or just revamped into something else ?

DzikuVx, 2018-07-27:
@c4xp use mixer to add correct rules. It will be in release notes

closed 2018-09-28 by teckel12