iNavFlight/inav #3473: 2.0.RC1 nav_extra_arming_safety review

Artyum, 2018-06-28:
Config: nav_extra_arming_safety = OFF
1. Wait for GPS Fix
2. Arm witch stick
3. Disconnect GPS (simulate hardware failure)
Result: There is no system message about GPS failure; the sats number indicator blikns. The coords are fixed at point of hardware failure.
4. Disarm with stick (OK)
5. Rearm with stick
Result: When "summary Screen" goes off, the system message shows "GPS Failure. Unable to arm." Arming is not possible.

shellixyz, 2018-06-28:
When you disconnected the GPS the sat number element blinked but how many sats was displayed ? Apart from that seems OK to me. I guess a system message about GPS failure could be added.

Artyum, 2018-06-28:
Dear @shellixyz, the sat number blink at last known value that was 5. A warning message would be nice but, most important is that disabled 'extra arming safety' is supposed to allow arm without gps. In this particular case, imho it doesn't work as expected.

Artyum, 2018-06-29:
I did another check, It seems that arming is not possible if GPS is not connected. According to docs: "nav_extra_arming_safety: If set to ON drone won't arm if no GPS fix and any navigation mode like RTH or POSHOLD is configured", so if set to OFF drone (and airplane?) will arm if no GPS fix but still the hardware must be connected (?). That way unfortunately GPS cannot be taken off in case of flying on sight.

stronnag, 2018-06-29:
Wrong. I do this on multiple models (no GPS, mag or baro, just fun).

Artyum, 2018-06-29:
I've just tested on 2.0.0.RC2. I can not arm if GPS is not connected, but I can arm immediately after connecting GPS (without fix)

stronnag, 2018-06-29:
Because ...

* it allows me to test inav on less conventional configurations (small tricopter,vtail)
* it allows me to test iNav on additional FC hardware
* betaflight support for tri, vtail is somewhat variable
* until recently, betaflight did not restore acc settings, making upgrade a pita
* as a long standing member of the iNav team, I'd prefer to support iNav (though I do admit to having a couple of βF machines.

Will that do?

teckel12, 2018-06-29:
@stronnag If my comment was taken as trollish, that was not the intent.  I get being a developer and testing the code part, but didn't know about the limited support for tri and vtail in Betaflight.  I'm satisfied with your answer, continue using INAV 😉

stronnag, 2018-06-29:
then back our usual programming ... 

@Artyum if you wish to progress this, you need to supply a modicum of information (as per the issue template). In particular:

* a CLI `diff`
* The output of the cli `status` command

Otherwise, we are, at best, guessing.

Artyum, 2018-06-30:
diff and status below. Seems that flag HWFAIL is set because of unavailable GPS despite nav_extra_arming_safety = OFF.

`# diff

# version
# INAV/OMNIBUSF4V3 2.0.0 Jun 29 2018 / 12:35:56 (a8415e89c)

# resources

# mixer
mmix 0  1.000  0.000  0.000  0.000

# servo mix
smix 0 3 0 100 0
smix 1 4 0 100 0
smix 2 5 2 100 0
smix 3 2 1 100 0

# servo
servo 2 1000 2000 1500 45 
servo 3 1000 2000 1500 50 
servo 4 1000 2000 1500 50 
servo 5 1000 2000 1500 -65 

# feature
feature THR_VBAT_COMP
feature MOTOR_STOP
feature GPS
feature RSSI_ADC
feature AIRMODE
feature PWM_OUTPUT_ENABLE

# beeper

# map

# serial
serial 0 0 115200 38400 0 115200
serial 2 2 115200 115200 0 115200
serial 5 64 115200 38400 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100
aux 1 3 4 1300 2100
aux 2 9 4 1700 2100
aux 3 8 3 1700 2100
aux 4 35 4 1300 1700
aux 5 10 4 900 1300
aux 6 27 6 1700 2100
aux 7 32 5 1300 1700
aux 8 33 5 1700 2100

# adjrange

# rxrange

# osd_layout
osd_layout 0 0 1 2 V
osd_layout 0 1 24 1 V
osd_layout 0 2 0 0 V
osd_layout 0 7 1 3 V
osd_layout 0 9 24 2 H
osd_layout 0 13 3 7 V
osd_layout 0 14 1 11 V
osd_layout 0 15 22 7 V
osd_layout 0 20 1 13 V
osd_layout 0 21 1 12 V
osd_layout 0 22 14 2 V
osd_layout 0 23 1 1 V
osd_layout 0 24 12 3 H
osd_layout 0 26 22 6 V
osd_layout 0 28 23 12 V
osd_layout 0 29 23 13 V
osd_layout 0 30 1 14 V
osd_layout 0 32 24 2 V
osd_layout 0 33 24 3 V
osd_layout 0 34 10 1 V
osd_layout 0 38 24 2 H
osd_layout 0 41 22 5 V
osd_layout 0 42 4 5 H
osd_layout 0 46 11 4 H
osd_layout 0 48 10 4 H
osd_layout 0 77 18 3 H
osd_layout 1 2 0 0 V
osd_layout 1 3 8 6 V
osd_layout 1 13 3 7 H
osd_layout 1 14 1 11 H
osd_layout 1 15 22 7 H
osd_layout 1 20 1 13 H
osd_layout 1 21 1 12 H
osd_layout 1 22 14 2 V
osd_layout 1 34 10 1 V
osd_layout 1 43 0 0 V
osd_layout 1 46 2 13 H
osd_layout 1 47 2 12 H
osd_layout 1 48 1 1 H
osd_layout 2 2 0 0 V
osd_layout 2 3 8 6 V
osd_layout 2 22 14 2 V
osd_layout 2 34 10 1 V
osd_layout 2 45 0 0 V

# master
set looptime = 2000
set gyro_hardware_lpf = 98HZ
set acc_hardware = MPU6000
set acczero_x = 69
set acczero_y = -53
set acczero_z = -299
set accgain_x = 4067
set accgain_y = 4077
set accgain_z = 3989
set mag_hardware = NONE
set baro_hardware = BMP280
set pitot_hardware = NONE
set receiver_type = SERIAL
set rssi_scale = 102
set serialrx_provider = SBUS
set rx_min_usec = 988
set rx_max_usec = 2012
set blackbox_rate_denom = 2
set max_throttle = 1950
set failsafe_delay = 15
set failsafe_throttle_low_delay = 0
set failsafe_procedure = RTH
set failsafe_min_distance_procedure = RTH
set platform_type = AIRPLANE
set model_preview_type = 14
set small_angle = 180
set switch_disarm_delay = 250
set gps_min_sats = 5
set inav_reset_home = FIRST_ARM
set nav_extra_arming_safety = OFF
set nav_wp_radius = 1000
set nav_manual_climb_rate = 500
set nav_min_rth_distance = 1000
set nav_rth_climb_first = OFF
set nav_rth_allow_landing = FS_ONLY
set nav_rth_alt_mode = FIXED
set nav_rth_altitude = 9000
set nav_fw_min_thr = 1250
set nav_fw_max_thr = 1800
set nav_fw_bank_angle = 45
set nav_fw_pitch2thr = 20
set osd_video_system = PAL
set osd_time_alarm = 15
set osd_alt_alarm = 0
set osd_dist_alarm = 0
set osd_neg_alt_alarm = 0
set osd_crosshairs_style = AIRCRAFT
set osd_left_sidebar_scroll = GROUND_SPEED
set osd_right_sidebar_scroll = ALTITUDE
set name = MENTOR

# profile
profile 1

set fw_p_pitch = 3
set fw_ff_pitch = 33
set fw_p_roll = 7
set fw_i_roll = 15
set fw_ff_roll = 68
set fw_p_yaw = 45
set fw_i_yaw = 5
set fw_ff_yaw = 15
set max_angle_inclination_rll = 600
set max_angle_inclination_pit = 600
set rc_expo = 60
set pitch_rate = 15
set yaw_rate = 9
set manual_rc_expo = 60

# battery_profile
battery_profile 1`

> # status
System Uptime: 378 seconds
Current Time: 0000-01-01T00:00:00.000+00:00
Voltage: 15.49V (1S battery - OK)
CPU Clock=168MHz, GYRO=MPU6000, ACC=MPU6000, BARO=BMP280
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=UNAVAILABLE
SD card: Manufacturer 0x2, 15114240kB, 06/2015, v2.0, 'SA16G'
Filesystem: Ready
Stack size: 6144, Stack address: 0x10010000, Heap available: 920
I2C Errors: 0, config size: 4129, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 2, used = ADC 2
      RSSI : configured = ADC 3, used = ADC 3
   CURRENT : configured = ADC 1, used = none
  AIRSPEED : configured = none, used = none
System load: 0, cycle time: 2001, PID rate: 499, RX rate: 55, System rate: 9
Arming disabled flags: HWFAIL CLI

stronnag, 2018-06-30:
Your GPS is not working (probably wrong connection), yet is enabled. The answer is facing you (i.e. the `HWFAIL` , ` GPS=UNAVAILABLE`).  Disable the GPS `FEATURE -GPS'. The craft will then arm.

stronnag, 2018-06-30:
Or fix the GPS so it detected as working.

digitalentity, 2018-06-30:
@Artyum 

Your issue description already explains this behavior:

>3. Disconnect GPS (simulate hardware failure)

Hardware failure ALWAYS blocks arming, regardless of other settings. Navigation Arming Safety setting assumes that GPS is working, but don't have a fix.

Closing as "working as intended"

closed 2018-06-30 by digitalentity