iNavFlight/inav #3474: Waypoint missions do not activate, when switching from manual to WP

wilco1967, 2018-06-28:
### Board and Version
INAV/MATEKF722 2.0.0 Jun 28 2018 / 17:42:06 (e0c0d5ce)
(self compiled from development)

### Behavior
Airplane (Mini Talon)
When flying, and loaded waypoint, when I switch immediately from manual mode to waypoints, the waypoint mode does not activate, and it says on OSD, it's in AIR mode (and so it behaves),

when coming from Air mode, or from level or from oshold, or from 3CRS mode, it works fine.

This is something similar, as the poshold issue 3448, that was solved earlier. 
BTW: issue 3448 works perfect now... Tried it in to same flight todays.... Thanks again !

For information, I use only a single RC channel for switching modes. And as a channel can only have one value at any one time, it shouldn't matter from where it was before.

in my case, manual is set up as 1000 uS on CH5
WP is set up as 1500 uS, (on the same CH5)

other modes I use are (all on channel 5)
1000 - manual
1100 - Angle mode
1200 - poshold
1300 - airmode
1400 - 3D cruise
1500 - Waypoint
2000 - RTH

The Taranis mixer determines the priority of different switches (last mixer line for CH5 takes priority over (i.e. overwrites) previous lines, before it is send out)... iNav only 'sees' the final channel value

So if I have conflicting switch positions, the Taranis logic decides what gets send 
I use voice messages triggering on final output value (uS), so I always hear what is actually being sent, in case I accidently have a switch in the wrong position).

In the modes tab in iNav, the required functions are activated, based on the requested channel value (Ch5).

Only when coming from 1000 (manual), going straight to 1500 (waypoints), waypoint fails to activate, and it says 'AIR' on OSD.. 
when coming from for example 1100 (level), or 1200 (poshold) or 1300 (airmode), going to 1500 (waypoints), it works perfectly (on OSD showing as AH for the mode, and a system message about on it's way to waypoint.)

here's the video with OSD overlay (switching starting almost at the beginning)
[https://drive.google.com/open?id=1Xg7HKFkzlkEv8M10L_FaSLq-p9uDJ9L1](url)

the CLI dump file
[https://drive.google.com/open?id=1q6Pzrya3L9OaZbbXZoaEMLeAtuk53J8z](url)

And finally the blackbox log (WP switching starting around 3 minutes in the log, for a few minutes.... first trying WP on/off, later poshold on/off, which works now perfect btw...  the rest is just random flight to finish off the battery ;-)
[https://drive.google.com/open?id=1qnQnehq5cOwOAlN254BCqfcgjlLeHV-1](url) 

It's not really an important issue.... it works usually very good.... so no rush to get it sorted...
Everything else seems to work fine (just a few details on the new OSD elements, but that's already being adressed as it seems....).

Thanks a lot !

giacomo892, 2018-06-28:
There is actually a check in the code that prevents to enable WP if you are on MANUAL mode.
What if you stop on ACRO and then switch to WP after a short period of time?

wilco1967, 2018-06-28:
I guess I could try that....
Just needs some delay on the Taranis side ...
Is there a minimum time it needs to be out of manual, before it will accept WP ?

giacomo892, 2018-06-29:
@wilco1967 Just switch slowly. And report back :)

wilco1967, 2018-06-29:
I know it works if I go through air mode first.... Already tried....
I understand now this is actually a feature for people who use different channels for different switches... Now I know, no need to fix anything... I almost never would switch from manual straight into WP anyway....

Thanks for the clarification !

closed 2018-06-29 by wilco1967

giacomo892, 2018-06-29:
@wilco1967 Don't worry. Just take your time when switching 🥇