iNavFlight/inav #3480: Compass headings off - N=0 E=70 S=140 W=280

onlydave, 2018-06-30:
### Board and Version
Omnibus f4 v5
INAV/OMNIBUSF4V3 1.9.1 Apr 21 2018 / 13:13:58 (03a5c1922)

### Behavior
* Toilet Bowling
* East, South and West degrees are off by 10 to 40 degrees

#### Blackbox logs/Config (.ZIP) where problem occurs
Dump on [PasteBin](https://pastebin.com/NZa6iryg)

Last flight [LOG00068.TXT.zip](https://github.com/iNavFlight/inav/files/2151670/LOG00068.TXT.zip)

I've run the config a few times and there's no change in behaviour each time, been working on this for days now. Still a bit of wobble in the pid tuning but it's flyable in angle mode with and without nav althold.

Any help would be much appreciated!

DzikuVx, 2018-07-03:
* improve calibration
* move further away from the frame, power cables, motors and anything else metalic

closed 2018-07-03 by DzikuVx

onlydave, 2018-07-05:
Strangest part is that according to the logs it's always facing 250 degrees when it's turned on. No mater it's orientation