iNavFlight/inav #3483: Omnibus F4 Pro servo output issue

gregbiv, 2018-06-30:
Hello everyone! I am a "happy" owner of Chinese omnibus f4 pro. I am trying to make my tricopter working with iNav.

What I have:
1. Servo pin attached to PWM6
2. Servo 5v connected to PDB
3. tri_unarmed_servo is set to ON

On 1.9.1 I see on the motor tab for motor 6 values are changing, however, the motor is not responding.

On 2.0.x By default on tricopter setup in mixer tab servo 6 is set to be YAW stabilizer but for some reason on motor tab motor 7 is active. Tried to set servo 5 and motor 6 was active but it didn't make servo working.

What else I've tried? It works on iNav 1.7.1 (since PWM values are hardcoded there). It works on Betaflight with resource remapping feature.

gregbiv, 2018-06-30:
Let me know if you need the dump for both version, but all values are default except radio and modes to test servo.

shellixyz, 2018-07-03:
Please post a configuration diff and specify the firmware version it is coming from

davidngrc, 2018-07-11:
Please flash iNav 1.9.1 and try my config on the second post here
https://github.com/iNavFlight/inav/issues/3558

note, I use another 5v bec to power my servo, only the servo signal wire is connected to Omnibus F4 Pro 
I somehow make it to work. (still don't know why it doesn't work before)

gregbiv, 2018-07-16:
sorry, it took time to get back to my tricopter

# version
# INAV/OMNIBUSF4PRO 2.0.0 Jul 14 2018 / 22:40:31 (f2cf315ad)

# reset configuration to default settings
defaults noreboot

# resources

# mixer
mmix 0  1.000  0.000  1.333  0.000
mmix 1  1.000 -1.000 -0.667  0.000
mmix 2  1.000  1.000 -0.667  0.000

# servo mix
smix 0 5 2 100 0

# servo

# feature
feature MOTOR_STOP
feature PWM_OUTPUT_ENABLE

# beeper

# map

# serial
serial 0 64 115200 38400 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1300 1700

# adjrange

# rxrange

# osd_layout

# master
set gyro_sync = ON
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 90
set gyro_notch1_hz = 400
set gyro_notch1_cutoff = 300
set gyro_notch2_hz = 200
set gyro_notch2_cutoff = 100
set acc_hardware = MPU6000
set acczero_x = 80
set acczero_y = -51
set acczero_z = 336
set accgain_x = 4101
set accgain_y = 4069
set accgain_z = 4044
set mag_hardware = HMC5883
set magzero_x = -840
set magzero_y = -426
set magzero_z = -1016
set baro_hardware = BMP280
set pitot_hardware = NONE
set receiver_type = SERIAL
set serialrx_provider = SBUS
set motor_pwm_rate = 490
set align_board_yaw = 1800
set platform_type = TRICOPTER
set model_preview_type = 1
set async_mode = GYRO

# profile
profile 1

set mc_p_pitch = 55
set mc_i_pitch = 40
set mc_d_pitch = 15
set mc_p_roll = 55
set mc_i_roll = 40
set mc_d_roll = 15
set mc_p_yaw = 90
set mc_i_yaw = 20
set dterm_lpf_hz = 80
set dterm_notch_hz = 260
set dterm_notch_cutoff = 160
set tpa_rate = 20
set tpa_breakpoint = 1650
set roll_rate = 70
set pitch_rate = 55
set yaw_rate = 25

# profile
profile 2

# profile
profile 3

# battery_profile
battery_profile 1

# battery_profile
battery_profile 2

# battery_profile
battery_profile 3

# restore original profile selection
profile 1
battery_profile 1

# save configuration
save

# 

tried today newest inav and configurator - same issue

gregbiv, 2018-07-16:
# version
# INAV/OMNIBUSF4PRO 1.9.1 Apr 21 2018 / 13:12:44 (03a5c1922)

# reset configuration to default settings
defaults noreboot

# resources

# mixer
mixer TRI

# servo mix

# servo

# feature
feature MOTOR_STOP
feature PWM_OUTPUT_ENABLE

# beeper

# map

# name

# serial
serial 0 64 115200 38400 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1300 1700

# adjrange

# rxrange

# master
set gyro_sync = ON
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 90
set gyro_notch1_hz = 400
set gyro_notch1_cutoff = 300
set gyro_notch2_hz = 200
set gyro_notch2_cutoff = 100
set acc_hardware = MPU6000
set acczero_x = 159
set acczero_y = -98
set acczero_z = 683
set accgain_x = 4102
set accgain_y = 4069
set accgain_z = 4044
set mag_hardware = HMC5883
set magzero_x = -867
set magzero_y = -425
set magzero_z = -1504
set baro_hardware = BMP280
set pitot_hardware = NONE
set receiver_type = SERIAL
set serialrx_provider = SBUS
set motor_pwm_rate = 490
set align_board_yaw = 1800
set async_mode = GYRO

# profile
profile 1

set mc_p_pitch = 55
set mc_i_pitch = 40
set mc_d_pitch = 15
set mc_p_roll = 55
set mc_i_roll = 40
set mc_d_roll = 15
set mc_p_yaw = 90
set mc_i_yaw = 20
set dterm_lpf_hz = 80
set dterm_notch_hz = 260
set dterm_notch_cutoff = 160
set tpa_rate = 20
set tpa_breakpoint = 1650
set roll_rate = 70
set pitch_rate = 55
set yaw_rate = 25

# profile
profile 2

# profile
profile 3

# restore original profile selection
profile 1

# save configuration
save

# 

same thing here

closed 2018-09-28 by stale[bot]