iNavFlight/inav #3487: iNav2.0.0-RC2: Batt voltage 'sag compensated'=no output on smartport. OmnibusF4V3

Ant73, 2018-07-01:
FC=Omnibus F4 V3 (pinout attached for board ID) - target OMNIBUSF4V3
Rx = R-XSR
Tx = Taranis X9D+
smartport connected via pwm6 using custom target.h (attached)

Raw battery voltage is correctly displayed on Tx. Setting 'sag compensated' on the config tab gives an output of 0v but Tx says it has a connection. (flashing *)
[image]

[target.docx](https://github.com/iNavFlight/inav/files/2152371/target.docx)

Ant73, 2018-07-01:
@teckel12 
Oh ok, thanks. I searched around for a while but couldn't find anything on it.
Yes, an alternate build might be an idea given resource mapping isn't on the cards. What I'd really love to see is if someone (much more knowledgeable than me!) could produce a wiki on how to do this properly and which pins are able to be reallocated. This build is for my quad but my plane is using pwm5/6 for servos so my only other options there are the LED or maybe Ch2, but I really don't know what Ch2 is for or if it is even possible to remap that for smartport.

teckel12, 2018-07-01:
As the Omnibus F4 is so popular, we already have a unique build for a relocated LED pin, and the PWM5 & 6 pins go unused for many builds, it seems logical that alternative targets would make sense.

The only issue I can think of is that we already have so many Omnibus F4 targets and so many different Omnibus F4 versions and model confusion, that creating more targets may just become even more complicated and difficult for the novice user to figure out. Personally, I only use the Omnibus F4 Pro which has soft serial pads which I have no problem soldering to. And for the next build I'll use FPort which can share a UART with telemetry. So I probably won't even need a soft serial port. So really, going forward, soft serial ports won't be as useful. And that may be the reason no one is that interested in going doing the deed with may cause more problems than it's worth. But who knows, maybe I'll get the need and do it some weekend.

shellixyz, 2018-07-03:
The sag compensated voltage is only available when you have a current sensor configured. Either real or virtual. I should add this to the documentation.

VoicOfReason, 2018-07-14:
I'm having the same issue with the SP Racing F4 Evo on iNav RC3. So, there is/will be no way to activate "sag" without a current sensor attached?

teckel12, 2018-07-14:
@VoicOfReason You could add a virtual current sensor as mentioned.

closed 2018-09-26 by stale[bot]