iNavFlight/inav #3488: Unexpected Disarm

kiryam, 2018-07-01:
I have strange behaviour with failsafe and inav 2.0 (46ef27db8) + SP Racing F3 + OrangeRX LRS
I have a video and blackbox data ( [failsafe.TXT](https://github.com/iNavFlight/inav/files/2152702/failsafe.TXT) 100%, 2 events,  see the last one).
I made video with combined blackbox and video.

https://youtu.be/kNTUjnrySFQ?t=1m43s
1. I reached out of signal zone
2. Controller get it and turn `RTL ON`
3. I have signal back, switched  `RTL OFF`,
4. Signal lost `RTL ON` 
5. `RTL OFF`,
6. I see on OSD signal about failsafe (Not about RTL. My receiver rather than stop to send signals, sends preset )
7. Aircraft does nothing to go home (I have set RTL on failsafe) 
8. I had `DISARM` (I have disarm switch and not touched them)

stronnag, 2018-07-01:
```
2018-07-01T11:31:29+0100 Set home 55.894159 37.250734 (LTM OFrame true)
2018-07-01T11:31:46+0100 New LTM Mode Altitude Hold (8) 1 17s 55.894322 37.250956 0 10
2018-07-01T11:32:26+0100 New LTM Mode Cruise (18) 1 57s 55.894081 37.250598 0 20
2018-07-01T11:33:26+0100 New LTM Mode Return Home (13) 1 117s 55.900756 37.241800 0 4
2018-07-01T11:33:32+0100 New LTM Mode Cruise (18) 1 123s 55.900512 37.243107 0 20
2018-07-01T11:33:47+0100 Disarm from LTM
2018-07-01T11:33:47+0100 Distance = 2020.9, max speed = 23.0 time = 138
2018-07-01T11:33:48+0100 BB Disarm Failsafe (6)
```
[image]

* purple = CRUISE
* yellow = RTH

kiryam, 2018-07-01:
@stronnag  looks like it first flight in blackbox 2 events, need the last one

stronnag, 2018-07-01:
The above is the last log, F/S disarm 600m from home.  I assumed it was the last as that has F/S events. Here's the mode changes for each:
```
[jrh@eeyore]$ inav_modes.rb ~/dl/failsafe.TXT 
iNav version = 2.0.0 (states eq 2.0.0)
Time(s)	Elapsed(s)	State
 234.1	(   0.0)	nav_state_launch_wait (26)
 245.1	(  11.0)	nav_state_launch_in_progress (28)
 250.1	(  16.0)	nav_state_idle (1)
 250.1	(  16.0)	nav_state_althold_in_progress (3)
 335.1	( 101.0)	nav_state_rth_head_home (10)
 335.2	( 101.1)	nav_state_althold_in_progress (3)
[jrh@eeyore]$ inav_modes.rb -i 2 ~/dl/failsafe.TXT 
iNav version = 2.0.0 (states eq 2.0.0)
Time(s)	Elapsed(s)	State
 102.4	(   0.0)	nav_state_launch_wait (26)
 115.3	(  13.0)	nav_state_launch_in_progress (28)
 118.0	(  15.6)	nav_state_idle (1)
 118.0	(  15.6)	nav_state_althold_in_progress (3)
 155.6	(  53.2)	nav_state_cruise_3d_in_progress (33)
 156.6	(  54.3)	nav_state_cruise_3d_adjusting (34)
 157.4	(  55.1)	nav_state_cruise_3d_in_progress (33)
 158.6	(  56.2)	nav_state_cruise_3d_adjusting (34)
 159.0	(  56.6)	nav_state_cruise_3d_in_progress (33)
 198.8	(  96.4)	nav_state_cruise_3d_adjusting (34)
 210.3	( 107.9)	F/S
 210.3	( 107.9)	nav_state_rth_head_home (10)
 216.3	( 113.9)	nav_state_cruise_3d_in_progress (33)
```

The first flight seems to end normally, at home, (disarm reason:4, switch), mainly in alt-hold mode, like this: 
[image]

Full 2nd flight 
[image]

kiryam, 2018-07-01:
Ahh, yes 2nd flight it is exact event on video.

stronnag, 2018-07-01:
So failsafe is still asserted when CRUISE takes over again after the RTH section, then it's the long walk. 

Did you move the sticks to go back to CRUISE, or was is the FC's choice?

kiryam, 2018-07-01:
@stronnag I am not, but I have failsafe preset in my RX, If signal lost he moves sticks to preset position, (ARM ON, RTL ON, ANGILE ON, CRUISE OFF) after I get signal back I think the mode was changed auto to my TX stick positions (ARM ON, RTL OFF, ANGILE ON, CRUISE ON)

kiryam, 2018-07-03:
I set switch to `FILESAFE ON`
1. `ARM` (arm icon displayed in configurator)
2. turn filesafe switch (**no arm icon** in configurator, filesafe icon on)
[disarm2.TXT](https://github.com/iNavFlight/inav/files/2161022/disarm2.TXT)

kiryam, 2018-07-03:
I have to updated to latest inav 2.0 RC2
I recorded video from configurator
https://youtu.be/DC-zUTYgLDA

kiryam, 2018-07-03:
Thats only happens if my throttle stick is zero. I found if I set throttle 1600 (>1500) All works as expected and aircraft not goes to disarm

kiryam, 2018-07-03:
So.  It is expected behaviour because of setting 
`failsafe_throttle_low_delay = 100`
[image]

Fixed by 
`failsafe_throttle_low_delay = 100`

junwoo091400, 2018-07-16:
@kiryam Um not related to this issue, but I can't get SPracingF3 to upgrade it's firmware and would appreciate if you can share how your process goes.

What setting in configurator do you use(Manual Baudrate, No reboot Sequence, etc.)?
Also, do you use boot pins?

I tried almost everything, but just couldn't get it done :( I hope you can give me a clue.

closed 2018-09-28 by stale[bot]