iNavFlight/inav #3489: Neo 8 GPS no data transmission

wookie666, 2018-07-01:
I have a neo 8m that when connected to u-center, outputs data fine, and locks on, I then connect it to my betaflight f4 flight controller using UART3, GPS Baud rate set to 115200 OR 38400, GPS setup in configuration. And i dont get any data transmitted in the GPS tab.

I have RX-TX and TX -RX correctly. all tested using u-center.

Uart6 is set to sbus,

This was working before, but for some reason now its not.

Ive also tried to flash the latest betaflight f4 config using inav, but for some reason, my board just wont flash, Ive been out of the scene for about 6 months, and cannot for the life of me get this to work.

Any ideas?

Gregmaan, 2018-07-03:
I have a different problem what may be related to this:
Min GPS Sat set to 7
With up to 10 sats I get no LAT/Lon values in Configurator or LUA telemetry.
Is there a hard coded GPS quality filter (HDOP, PDOP)?

shellixyz, 2018-07-03:
@wookie666 Try to reset your module configurations to defaults
@Gregmaan If you got `nav_extra_arming_safety` set to `OFF` it is normal. You can always arm but if no GPS fix then HOME is not set.

teckel12, 2018-07-03:
@Gregmaan After you get 7+ sats, from your transmitter do a `discover telemetry sensors`.  You may not be seeing LAT/LON in LuaTelemetry if the sensors don't exist, and they don't exist until there's a GPS lock.  See: https://github.com/iNavFlight/LuaTelemetry#from-transmitter for more information on setting up the telemetry sensors for LuaTelemetry.

Gregmaan, 2018-07-03:
@shellixyz I did not set nav_extra_arming_safety to OFF. I had 10 sats but no GPS position with 10 sats there should be a 3d lock in every case. 3 minutes later I had lat/lon data also with 10 sats.
@teckel12 after about 3 min with 10 sats suddenly the data are there, but as I told before there should be a lock long time before. Maybe it´s only a problem with the 3D lock. I set up the telemetry exactly as described in your recommended link. Now I know what happened when I got the message "sensor lost" ...
I will resetup telemetry sensors with 3D lock and see if this makes a differnce.
Many Thanks!

wookie666, 2018-07-03:
Thanks Shellie. I don't reset them last night using ucenter and it connected straight up. Weird because this GPS was working before and has sit still for 6 months

closed 2018-07-03 by wookie666

teckel12, 2018-07-03:
@Gregmaan There's no hard-coded filter for a HDOP of >2 in Lua Telemetry.  It's not till the next release of Lua Telemetry that it will even know what the HDOP is.

Gregmaan, 2018-07-04:
@teckel12 When I connect the FC to the configurator I can see HDOP values. If the HDOP is >2 I get a warning on my RC ..Bad GPS and the Lat/lon is blinking(Or before first arm I see no GPS data). If the HDOP is <2 I got the message GPS good and I can see GPS lat/lon.

teckel12, 2018-07-04:
Nope, LuaTelemetry 1.3.1 thinks GPS is good when INAV sends it information that it has a GPS lock, home lock, and at least 6 satellites. The next release of LuaTelemetry changes this to only look at the GPS fix, basically giving even more control to INAV to determine if there's a GPS lock or not.

If there's logic of GPS lock based on HDOP, it's in INAV. But, I know it's not a HDOP of 2.0 as I get a GPS fix at 4.0, if not higher (that's just the highest HDOP I remember).

stronnag, 2018-07-04:
In iNav, GPS lock status is that reported by the GPS (PVY / nav solution).

teckel12, 2018-07-04:
And INAV just passes that along via telemetry which LuaTelemetry reports as either a GPS lock or GPS lost.