iNavFlight/inav #3491: Low Power mode (low current consumption)

dzing63, 2018-07-01:
Hi everybody, this is a request for a "Low Power mode" switch.
INAV is the firmware more suited for airplanes, and it would be interesting if a low power consumption option is implemented. 

We can now find extremely low weight controllers (2 grams and under) with F3 or F4 cpus .... but the consumption is high and the weight then comes from batteries!
Typically you can expect around 100mA of consumption when running a F3 flight controller (72 Mhz) on a plane. 100mA is marginally influential when considering a medium/big drone, when most of the power goes into motors, or even in medium / big airplanes (when you have maybe 2000 maH batteries).

But an extremely interesting usege would be in indoor/lightweight planes, or even in gliders.
In this case the scenario is different. I'm developing small planes of about 30 to 40 grams weight, powered by one 1s 150maH battery. In this case 100mA is a HUGE consumption. The same is for
gliders, where there's no motor and the rx and servos could need maybe 25 to 30mA at idle.

Now the main problem is that in planes the CPU is used only af a few % (1 to 2% I see),
It would be interesting to be able to drastically LOWER the mA consumption of the controller.
I tried with the cpu_underclock variable and it works ... the CPU goes at 48 Mhz, the load is 5%, I get about 80mA of consumption. But that'n not enough.

I read the discussion about cpu_underclock, the USB limit of 48 Mhz, examined the STM32F3 datasheet and also examined the source code, coming to the following conclusions:
- current is proportional to speed: we could lower current to only 10 - 20 mA if we lower the speed enough
- if we go under 48 Mhz the USB will not work.... but FC control is still possible using a serial!
- we could also avoid the use of USB ... saving even more current
- even if we go to 8 or 16 Mhz the CPU load would remain under control (maybe 30%)

How about a special switch for "low power planes" (indoor / gliders) ? 
You'd make all the indoor / glider users happy!
I'm going to try this myself creating a custom firmware, but it would of course be better if the feature becomes standard in INAV!

Thank you!

digitalentity, 2018-07-01:
Designing low-power systems starts with hardware. No manufacturer is really concerned about power consumption of the board. Also, CPU core is not the main consumer - it's the CPU peripherals that draw most power - timers, SPI controller etc.

F4 CPU alone consumes about 80mA on 168MHz of which CPU core is only 40mA. Gyro adds another 5-7mA. Add leak currents in cheap BECs and LDOs and poorly designed board schematics and you easily get over 100mA.

Lowering the CPU clock won't help much. In your F3@48MHz example (total consumption of 80mA) CPU consumes only some 30-35mA, the remaining 50mA come from stuff external to CPU. Also, for the record, USB module consumes only ~1mA.

Even if we drop CPU speed to ridiculous 16MHz we'll only save about 15-20mA but it would require quite a bit of work to change all the clock divisors in all the drivers to ensure everything works as needed.

Redshifft, 2018-07-01:
Yeah, clock speed alone will not proportionally reduce power, we need to start reducing core voltages. STM32 mcu's seem to have power modes but good luck trying to make much sense of them past changing basic clocks.

teckel12, 2018-07-01:
Having built uber-low current systems designed to last several years with a single set of batteries, I can confirm that it starts with the hardware. If I wanted to build a super low powered flight controller, I'd probably consider a different  microcontroller and then select each component for power usage. Even components like voltage regulators and how switches and LEDs are configured is important.

In any case, having a lot of knowledge on this topic, I'd say it's well outside the scope of the FC software. A FC would need to be designed for this task.

Redshifft, 2018-07-01:
For sure, even the F7's are 10 years behind people like Intel in the silicon fab, but cheap and plentiful is always going to win for our average users :)

dzing63, 2018-07-09:
Well, I understand all your comments, for most people things are OK as they are, yet for using INAV with small batteries lowering the CPU frequency can be useful.

For those who are interested .... I made my own tests, and modified my copy of omnibus firmware so that instead of going to 48 Mhz when enabling cpu_underclock it lowers the PLL multiplier to 2 (instead of 6) so that I get 16 Mhz cpu clock. Of course I now have to control the FC with a serial cable instead of USB.

With a generic airplane preset at 16 Mhz (and no wait state for the flash) the cpu % is still a good 9%. The current consumption drops from 110 mA to 48-49. If you use a 150 maH battery that's a good improvement with no collateral effects from what I see, and it costs nothing.

teckel12, 2018-07-09:
@dzing63 I'd also suggest removing the LEDs that stay on from the FC, that's a trick to convert an Arduino Pro Mini to be fairly low current (along with software that goes into sleep mode).  The voltage regulator is the next thing to look at.  Removing the voltage regulator and replacing it with a higher efficiency component can also save quite a bit.  You can test what it will save by wiring in a voltage regulator on the output side of the on-board regulator.

Was the 110mA at full speed or at 48 Mhz?  The loss of USB control is an issue as most computers don't have a serial port anymore, effectively bricking the FC for many.

closed 2018-09-21 by stale[bot]

fiam, 2019-05-22:
@dzing63 I've been thinking about this lately. I think the best approach for achieving very low current consumptions would be to design an L4 based FC. A L433 at 80MHz with all peripherals enabled consumes around 18ma, which is 50-60ma less than an F303 at 72MHz also with all peripherals turned on. However, the FC wouldn't need to enable absolutely all the hardware, so maybe a realistic consumption for the MCU would be around 12ma (core itself needs 8.5ma).

Designing all the flight controller carefully could definitely reduce power usage by a lot. Thing is, I'm not sure there's a big enough market to justify the required R&D.

On the other hand, there are receivers with stabilisation from several vendors already on the market, which I'd assume it's the main feature you'd like to use from INAV (I don't think you're running an FPV setup with a 150mah battery). No idea about how much current they consume, but maybe they would be a better fit for this use case.